erpnet/ErpNet.FP #1: Selection of multi platform .Net Core serial port library

dekiland, 2019-03-12:
We must select **multi platform** .net core serial port library, which will be used as standard underlying communication with Fiscal Printers and Devices. This is team task, so I assigned this task to all of us. Please consider that there are no so many as NuGet packages, as well as we must select the **library with minimal dependencies and maximum portability**.
**Without this** library we will be forced to use only Microsoft Windows **System.IO.Ports**, which is not supported in .Net Core, **and stick only with Windows** :(

dilyanrusev, 2019-03-12:
Quick search resulted in https://www.nuget.org/packages/Mono.Posix.NETStandard/1.0.0-beta2 - https://github.com/mono/mono/tree/master/mcs/class/Mono.Posix

dilyanrusev, 2019-03-12:
Using mono POSIX, etc, we can take the idea from https://github.com/node-serialport/node-serialport/tree/master/packages/binding-abstract and https://github.com/node-serialport/node-serialport/tree/master/packages/bindings and do the work ourselves. Or, we could take the easy route and use the drivers provided by the Fiscal printer drivers. Some of them are in node and therefore work on linux.

dekiland, 2019-03-12:
Mono.Posix.NETStandard, does not implement System.IO.Ports, consider this: https://www.nuget.org/packages/MonoSerialPort/   (.NETStandard 2.0)

dilyanrusev, 2019-03-14:
Looks like somebody's already done the work for us. Looks great!

dekiland, 2019-03-18:
Ok, so issue is considered closed then.

closed 2019-03-18 by dekiland