erpnet/ErpNet.FP #8: Keep printer time synced with NTP

i-argentinski, 2019-05-03:
Keep the printer synchronized with EU NTP time server.

The service should regularly check (or on each PrintReceipt) NTP server and if the printer current time is out of sync - set it.

If the difference between the NTP and the FP is more than 30 sec - then set the printer time.

The NTP server pool should default to some EU NTP server pool. But the address of the NTP server should be configurable through the config file.

i-argentinski, 2019-05-03:
The syncing should be performed:
- Upon server start
- Every hour after this

dekiland, 2019-05-09:
We will release new enhancement, Issue #9, setdatetime functionality, and will return current fiscal device datetime on status request. So this issue will be closed.