erpnet/ErpNet.FP #13: Support async printing and queueing

i-argentinski, 2019-05-16:
The idea is that each printing request will be enqueued in a queue.

(1) /printreceipt
will:
- enqueue the printing task.
- wait up to 29 secs for the queued task.
- return the usual PrintInfo result

(2) /printreceipt?asynctimeout=15000
will:
- enqueue the printing task.
- wait up to 15000 ms. 
- If the print task is finished within 15000 ms - return the usual PrintInfo result
- if the print job took more than 15000 ms - return TaskId result.

(3) /printreceipt?asynctimeout=0
will:
- enqueue the printing task.
- return TaskId result.

i-argentinski, 2019-05-16:
(4) /gettaskinfo?taskId=.....
returns TaskInfo

TaskInfo:
- TaskStatus:
-- Unknown
-- Enqueued
-- Running
-- Finished (if it has finished within the last 20 minutes)
- Result
The result should be in the same format, as the usual result of the async function.

When the server returns info (Finished) for a task, the info about the task is deleted from the server. Further queries about the status of the task will return (Unknown).

i-argentinski, 2019-05-16:
(5) /printreceipt?timeout=30000
This should limit the total timeout which the user is ready to wait for the print job (async or not).

(6) This could be combined with asyncTimeout. For example:
/printreceipt?asyncTimeout=5000&timeout=30000

This means:
- After 5000 ms, convert the request to async.
- After 30000 ms, if the job is not done, cancel it.

i-argentinski, 2019-05-16:
(7) The asyncTimeout and timeout parameters should be supported by:
- /printreceipt
- /printreversalreceipt
- /printdeposit
- /printwithdraw
- /printzerioingreport

g-vvv, 2019-05-20:
asyncTimeout is ready, timeout will be subject of new issue.

closed 2019-05-20 by g-vvv