erpnet/ErpNet.FP #25: Create Admin panel with MQTT activation

i-argentinski, 2019-05-28:
The admin panel would allow remote administration (making changes to the config file, etc.) of the print server.

Currrently, the admin panel would be a web page. In the future, Api access should also be allowed.

All admin access (web page AND api access) should be secured with a simple "Admin Secret". The admin secret is a password, stored in the config file. 

NOTE: If the admin password is NOT set, ALL admin access is DISABLED. E.g., there SHOULD be an admin password in order to access the admin part.

(1) Admin password
The password would be stored in the ErpNet.FP section, under the name "AdminPassword". It would be plain-text "secret" text.

NOTE: Obviously, this is not the highest level of security, but it is secure enough for remote access.

(2) MQTT secret
The MQTT secret is used to create topic names for the MQTT communication. It is stored in the config file, in the ErpNet.FP section, under the name "MqttSecret".

(3) Functions of the admin web page
**Location**: /admin/
**Login**: Each access of any admin page should require a login, requiring the user to type the admin password.
**Functions**:
* Button "Set Admin Password" - should allow the user to change the admin password.
* Button "Enable/Disable MQTT"
* If there is no MqttSecret, a new one is generated
* Upon MQTT activation (or just always), a JavaScript snippet should be displayed, showing the necessary JavaScript code to access the printing functions.
NOTE: The snippet should be with the MQTT secret already integrated into it.

i-argentinski, 2019-05-28:
(4) Under this same issue, implement MQTT remote control
(4.1) The topic name should include the MqttSecret

i-argentinski, 2019-05-28:
(5) MQTT Server
There should be "MqttServer" config option in the ErpNet.FP section.

The admin panel should allow the user to set the MqttServer.

NOTE: The MqttServer should also affect the JavaScript snippet.

g-vvv, 2019-06-20:
Admin panel is created, but MQTT support is postponed for future. May be there will be another issue about MQTT.

closed 2019-06-20 by g-vvv