erpnet/ErpNet.FP #29: Get the status of the last fiscal transaction

voltmachine, 2019-06-02:
---
name: Feature request
about: Suggest an idea for this project
title: 'Get the status of the last fiscal transaction'
labels: ''
assignees: ''

---

**Is your feature request related to a problem? Please describe.**
Frequently, there is a loss of connection with the software during printing. The ability to check the status of the last fiscal transaction any time after, will help to decide how to complete the sale.

**Describe the solution you'd like**
To have command which return the info of the last transaction. For example:
GET: /printers/dt525860/lasttransactionstatus

**Describe alternatives you've considered**
This is partially accomplished with asynchronous tasks, but it can not cover all cases.

g-vvv, 2019-06-02:
Because the library can be used from multiple clients at the same time, this is pointless and it is not transactional safe.

closed 2019-06-02 by g-vvv

i-argentinski, 2019-06-02:
The print server automatically requests the last status after each print job.

The status is contained in the task information.

Use the asyncTimeout parameter to force async execution. Use the taskId to obtain the last status of any print job.

NOTE: Currently, the task information is kept for 20 minutes after completing the job.