erpnet/ErpNet.FP #32: On error, result is "ok":true.

voltmachine, 2019-06-03:
When message of type error exists, the print is canceled corectlly, but result "ok" is set to true.
Is not it expected to be false?

```
{
   "taskStatus":"finished",
   "result":{
      "receiptNumber":"",
      "receiptDateTime":"0001-01-01T00:00:00",
      "receiptAmount":0.0,
      "fiscalMemorySerialNumber":"02746009",
      "ok":true,
      "messages":[
         {
            "type":"error",
            "code":"E406",
            "text":"Payment type Check unsupported"
         },
         {
            "type":"info",
            "code":"",
            "text":"Error occured while printing receipt items"
         }
      ]
   }
}}
```

g-vvv, 2019-06-03:
Thank you for your report. Please provide debug.log, to investigate your case.

voltmachine, 2019-06-03:
[debug.log](https://github.com/erpnet/ErpNet.FP/files/3248718/debug.log)

g-vvv, 2019-06-03:
Thank you, I make another build with simplified Ok property. You can try it, because I do not have the test devices @home. If it is OK, I will close the Bug. If the bug is reproduced, then tomorrow will continue with your case.

voltmachine, 2019-06-03:
Thank you.
I have tested it, it is still "ok":true:
[debug.log](https://github.com/erpnet/ErpNet.FP/files/3248809/debug.log)

g-vvv, 2019-06-03:
With the new build, I think the bug is resolved. You can try it. The problem was in initialization of descendants of the class DeviceStatus with copy constructor, without copying the Ok property. Thank you again for your feedback and have a nice evening.

voltmachine, 2019-06-03:
I confirm, it is fixed.
Thank you.

closed 2019-06-03 by voltmachine