erpnet/ErpNet.FP #36: Get a unique workplace identifier

voltmachine, 2019-06-07:
Get a unique workplace identifier on the device where the ErpNet.FP server is running.
By government requirements for workplaces, from annex 29 points 5, 8, 18, 18.1, 18.9.

It can also help for control fiscal devices usage - set defaults or restrict, when more than one available for a workplace.

For softwares without locally installed modules, it will eliminate the need to install an extra server to obtain workplace id.

Identifier may be a bios serial, drive serial, ethernet adapter or other unique identifier.
Response may be a hash, to keep a fixed length.

Example Request
GET: `/deviceuid`

Example Response
```
{
	"deviceuid": "2C290D367A8020BF6F75F5AB4B0E8F30"
}
```

i-argentinski, 2019-06-07:
We didn't find and such requirement in N-18, Annex 29, p.5, 8, 18.1 and 18.9.

The only requirement is to have unique fiscal printer id. This is returned by 
* /printers - list of all printers and their serial numbers
* /printers/xxx - information for specific printer

voltmachine, 2019-06-07:
As I understand workplaces and fiscal printers devices are different nomeclatures.
One workplace (pay desc which collect cash) can print from different printers. 

[image]

g-vvv, 2019-06-08:
Added /vars

closed 2019-06-08 by g-vvv