erpnet/ErpNet.FP #37: Customization of fiscal bon

palex, 2019-06-08:
**View of bon**
Would be great if all settings of OpenFiscalBon are presented into input JSON. For example Tremol's OptionReceiptFormat and OptionPrintVAT, for Datecs TillNumber etc.

Often I print additional info for specific articles (as serial numbers, contract number etc.), so in my subsys I have PrintBefore and PrintAfter in Items object. Additionally in PrintBefore and PrintAfter sometimes I print barcodes too.

g-vvv, 2019-06-08:
There are pretty good suggestions in your contribution, that will be useful to many people, such as headers, footers, barcodes. If they are not supported by а specific device, we will put some suitable alternatives. Thank you for your collaboration, it will be implemented.

g-vvv, 2019-06-08:
This will be an optional functionality that will not be supported in all cases, for example for software fiscalisation, these capabilities are not applicable, but are useful when the library is used in the context of physical fiscal devices. We will assume that these opportunities will be used for purely decorative purposes of the printed cash receipt.

closed 2019-06-20 by g-vvv