erpnet/ErpNet.FP #38: Implement Invoice, Credit and Debit notice functionality

moynzzz, 2019-06-11:
**Is your feature request related to a problem? Please describe.**
A lot of companies print the invoice in the fiscal bon. For example Tehnomarket in Bulgaria.

**Describe the solution you'd like**
I would expect an API endpoints where a request can make the fiscal printer to output a fiscal bon with invoice, credit or debit notice details on it.

**Describe alternatives you've considered**
The only alternative is to print an invoice or the other types of documents standalone and attach the fiscal bon to the document.

i-argentinski, 2019-06-11:
Actually, you CAN print invoices even now. Even invoices according to the legislation of many different countries.

We advise AGAINST using the fiscal printers built-in functionalities for invoices. 

You can use COMMENT lines to print invoice, following the legislation and the special requirements of each client.

Currently, we advise using COMMENT lines even for the line items. Of course, at least one fiscal line for the total amount is required.

The law does not force you to use the built-in functionalities of the printers.

closed 2019-06-11 by i-argentinski

moynzzz, 2019-06-12:
Oh I didn't thought about it this way. The COMMENT solution fits my cases perfectly. Thank you @i-argentinski