erpnet/ErpNet.FP #39: Cash amount and Timestamp in Info Method

palex, 2019-06-12:
**A few recommendations**
It's convenient to return in info method (which I use like an a ping or init) current cash because Cash Out, sometimes Storno, Operator Name change (of Datecs) and probably others will fail if application does not catch cash availability. Also current time of ECR will be valuable in info, so application can try to fix wrong calendar.
I found that devices fails if item price is zero (gift) and in my application I replace "Sell" method with a "print comment" if price is zero, but this behavior is constant and could be catch in the communication layer to avoid additional logic in application. 
And last, because subtotal have to be positive in every moment some negative items should be shifted at the end if current amount goes negative.

g-vvv, 2019-06-12:
Timestamp of ECR is available in every GET /status call.

g-vvv, 2019-06-20:
There is pre-validation for negative amount, payments, etc., before sending the commands to the fiscal device.

palex, 2019-06-20:
Great job g-vvv, 
Thanks for so important improvements, but I still can't show cash amount of user for equality check of physical cash and ECR cash. The only way remain XReport (20cm paper for only one important digit). Most of my customers are making cash check every hour because tax administration wants exact match between physical cash and device (you know it). It's convenient to show this amount in software without printing anything for quick check and eventually withdraw deference's (because of tips or mistakes in change). 

I imagine something like:
```
        public virtual (string, DeviceStatus) GetRawDeviceInfo()
        {
            var (responseFD, _) = Request(CommandReadFDNumbers);
            var (responseV, deviceStatus) = Request(CommandVersion);
            var (responseR, _) = Request(CommandReadRegisters,"0");
            return (responseV + responseFD + responseR, deviceStatus);
        }

       ....
            var info = new DeviceInfo
            {
                SerialNumber = serialNumber,
                FiscalMemorySerialNumber = fields[6],
                Model = model,
                FirmwareVersion = fields[4].Trim(),
                Company = "Tremol",
                CommentTextMaxLength = 30,
                ItemTextMaxLength = 32,
                OperatorPasswordMaxLength = 6,
                CashAmount = fields[7].Trim()
            };
      ....

g-vvv, 2019-06-20:
Reopening issue for ECR total cash amount available

closed 2019-06-28 by g-vvv