erpnet/ErpNet.FP #40: Payments mapping

palex, 2019-06-17:
**Payments mapping is hard coded**
Current defaults cover only device defaults but companies can change payment name on every payment type. So we need some kind of setup of payments (probably in appsettings.json) for GetPaymentTypeText to avoid resetting of devices.

**My suggestion is new optional object in appsettings under printers items**
Example:
```json
    "Printers" :  {
      "dt769373" :  {
        "Uri" : "bg.dt.p.isl.com://COM3"
        "PaymentTypes" :  {
          "card" :  "N",
          "cash" : "P",
          "packaging" : "I",
          "check" :  "C"
        }
      },
      "zk125928" :  {
        "Uri" : "bg.zk.zfp.com://COM2",
        "PaymentTypes" : {
          "card" : "7",
          "check" : "3",
          "cash" : "0",
          "packaging" : "4",
          "loyalty" : "2"
        }
      }
    }
```

palex, 2019-06-17:
It was a bad idea in context of NRA XML

closed 2019-06-17 by palex

wqweto, 2019-06-18:
Какво се има предвид под `NRA XML`? (Този космат английски в issues ме убива!)

Принципно [`enum PaymentType`](https://github.com/erpnet/ErpNet.FP/blob/master/ErpNet.FP.Core/Core/Payment.cs#L7) е доста тясно дизайнато, като се има предвид, че старите модели ФУ имаха по 8, а новите модели имат по 12 вида плащания -- `Талони`, `В.талони` и т.н. липсват в момента в енумерацията.

Идея нямам как може да се дизайне хем да има предефинирани типове като `Cash` и `Card` и едновременно да може да се разширява спрямо текущия модел на ФУ (още 8 типа hardcode-нати спрямо модела) и също спрямо текущите настройки на ФУ за настройваемите типове `Резервиран 1` и `Резервиран 2`.

От настройваемите един най-често се слага на `Ваучери`, защото по някаква причина дори в последните модели с 12 плащания липсват ваучери за храна. Със сигурност *никъде* не съм видял някой да плати с чек, но тази глупост продължава да се поддържа дори от последните модели, а едно време Датекс FP-3530 имаше отделно `Дебитна карта` и `Кредитна карта` дори.

Значи имайте предвид че писачите на firmware на ФУ не са the sharpest tool in the shed. . . по принцип. Примерно последните модели имат вид плащане `Сервиз`. . . Really? Вид плащане!?

i-argentinski, 2019-06-18:
Възможността за настройване на payment type и защо не трябва да е настройваем явно си плаче за първи въпрос във FAQ. Целта е да се програмира срещу стандарта. И от двете посоки - и като потребили и като създатели на принтерски драйвери. А стандартът определя не само трансфера на данни, но и _семантиката_ на стойностите.

Затова PaymentType е нарочно с малко стойности засега. Полека-лека ще се разширява списъкът със стойности, но само след като сме напълно убедени, че дадена стойност е необходима и стандартна.

Целта на JSON формата не е само отпечатване на принтер. Целта включва и:
- Обмен на данни
- Възможност за печат на един и същи файл на различни принтери (например при high available конфигурация)
- Възможност за употреба при софтуерна фискализация. Включително при вариант, при който един и същи JSON текст може да се отпечата както на физически, така и на виртуален ФП.

Ако липсва някой необходим тип конкретно кажете кой (или направо правете pull request) и може да го добавим.

Отделно, винаги стои възможността за fork. Но все пак, ако форк-вате, имайте предвид идеите, които описах по-горе.

i-argentinski, 2019-06-18:
И да, софтуерна фискализация още няма. Не е ясно дали ще има. И ако има - дали ще е на принципа на Виртуални ФУ. Но това е идеята.

wqweto, 2019-06-18:
Да, може би е добре да опишете или дори да подискутираме преди това. Не би трябвало да има семантика относно видове плащане точно в тази middleware услуга. Примерно има ли семантика във валутата на плащането в момента? Някакви обороти в лева или евро да калкулирате? По-скоро не ми се струва да е така.

По същия начин вида плащане (както валутата на плащане) е по-добре да останат "договорка" между клиентското приложение и ФУ като последното смята/печата тотали а ПОС системата разбира какво значи `Cash` и `Card` от гледна точка на НАП примерно.

Middleware е добре да може да върне списък с видовете за да може да се експонират в настройки на ПОС, защото в ПОС списъка с плащания които се предлагат на операторът първо не е 1:1 с този на ФУ, второ има данни които няма как да дойдат от ФУ -- примерно в ПОС може да се настройват някакви основания за плащане, така че различните видове плащания да се осчетоводяват различно, или примерно различните видове плащания да влизат в различни каси на backend.

Така че настройките на видове плащане може да са доста сложни, като в ПОС може да има и нужда от два вида плащане да се map-ват към един вид плащане във ФУ, но влизат в backend с различни основания или в различни каси. Примерно някакви фирмени съфинансирания за служителите, където X лева на ден ги поема работодателя (не се плащат нито кеш нито с карта а с отделен вид плащане) и прочее щуротии със сложна отчетност.

--
Със сигурност не става дума за fork, най-малкото проектът е в прекалено "рохко" състояние. Има смисъл от fork ако го ползваме в production и по някаква причина не приемате PR, а аз не съм сигурен дали *вие* самите го ползвате в production дори.

g-vvv, 2019-06-18:
Всеки ФП/КА изпраща към нап XML по спецификация подадена от НАП. Ето един
фрагмент от нея:

В брой

SCash

С чек

SChecks

Талони

ST

Сума по външни талони

SOT

Сума по амбалаж

SP

Сума по вътрешно обслужване

SSelf

Сума по повреди

SDmg

Сума по кредитни/ дебитни карти

SCards

Сума по банкови трансфери

SW

Резерв 1

SR1

Резерв 2

SR2

Във ФП/КА има твърда релация между стринга, който се подава като тип на
плащане и стринга, който се пуска в XML файла на НАП. Той е хард-коднат във
фърмуера на ФП/КА и не може да е промени чрез настройки, нито да се извади
с команда. Стринговете за тип на плащане на ФП/КА са подмножество на
възможните стрингове в XML спецификацията на НАП. Понякога даже няколко
стринга за тип на плащана на ФП/КА сочат към един стринг в XML
спецификацията на НАП.
Би било перфектно ако и ФП/КА не правеха ретранслация на стринговете, а
можеше да им подаваме направо стринговете от XML спецификацията на НАП.
Това би било възможно например чрез софтуерната фискализация, защото тогава
проблеми като "обратна съвместимост", няма да съществуват. Този проблем
всъщност е основния, при ФП/КА, заради който има такава ретранслация. По
вашата логика би било чудесно POS/СУПТО да може директно да използва XML от
списъка на НАП и да го пуска през библиотеката, той да минава без
транслация през ФП/КА и така да отиде в НАП. Но за съжаление, нещата не са
стандартизирани, а всеки си мисли някакви спецификации. Иван Аржентински се
опитва да наложи стандарти, даже с тази библиотека, стандарти, които да са
валидни и в ляво от нея POS/СУПТО и в дясно -> ХФУ/ВФУ и НАП.

wqweto, 2019-06-18:
Ооо, това е супер ценна информация, защото тогава стойностите на `enum PaymentType` могат да са 1:1 с валидните стойности от XML спецификацията без това да е "bottleneck" за клиентските ПОС системи.

Така ще може ПОС да си map-нат собствените видове плащания директно към стойностите от XML спецификацията през елементите на `enum PaymentType` енумерацията, а вече от там как се "промушват" без загуба през ФУ си е божа работа и въпрос на конфигурация предполагам.

Но още от сега виждам как примерно Incotex ще са проблемни с тяхната поддръжка на едва 5 вида плащания, щото по-горе броя 11 стойности в XML спецификацията. . .

g-vvv, 2019-06-18:
Точно там е основният проблем, Daisy, Incotex, старите протоколи на Datecs и старите на Tremol поддържат между 3 и 5 вида плащания, които са съпоставими с тези в XML спецификацията на НАП. Затова, за да може да се използва универсално библиотеката и да не се интересува потребителят и какъв ФП/КА стои отсреща, а да печатим на първия срещнат, ако например имаме пул от 5 принтера закачени на една машина, то е хубаво да се стандартизират нещата. Ето затова имаме enum, който със сигурност можем да поддържаме, независимо от типа на ФП/КА. Трябва да знаете и че понякога няколко стринга от ФП/КА сочат към един стринг на НАП. Т.е. няма пряка зависимост. Другата причина да не изберем списъка на НАП е, че е твърде аматьорски измислен, че да стане стандарт. Направен е без стратегически поглед върху възможните реални ситуации, в които се прави плащане.

g-vvv, 2019-06-18:
[XML Схеми на НАП](https://nap.bg/page?id=674)
По-специално внимание на xtask.xsd, nracommon.xsd

wqweto, 2019-06-18:
Мерси за линка!

(В `req31.xsd`  гледам `` и се смея с глас:-))