erpnet/ErpNet.FP #41: DATECS FP-800 over tcp failed

persuader, 2019-06-19:
Hi.
I have problem connecting to datecs fp-800 printer over tcp.

This is the console log.

info: ErpNet.FP.Server.Program[0]
      Starting the service...
info: ErpNet.FP.Server.Contexts.PrintersControllerContext[0]
      Detecting configured printers...
info: ErpNet.FP.Server.Contexts.PrintersControllerContext[0]
      Trying dt315421: bg.dt.p.isl.com://COM1, OK
info: ErpNet.FP.Server.Contexts.PrintersControllerContext[0]
      Trying dt315421_1: bg.dt.p.isl.tcp://192.168.1.94, failed
info: ErpNet.FP.Server.Contexts.PrintersControllerContext[0]
      Detecting done. Found 1 available printer(s).
Hosting environment: Production
Content root path: c:\Users\loboda\Downloads\win-x64
infoNow listening on: http://0.0.0.0:8001
Application started. Press Ctrl+C to shut down.
: ErpNet.FP.Server.Services.KeepAliveHostedService[0]
      Keep Alive Background Service is starting.
info: ErpNet.FP.Server.Services.KeepAliveHostedService[0]
      Keep Alive Background Service running...
info: ErpNet.FP.Server.Services.KeepAliveHostedService[0]
      Keep Alive Background Service done.
info: ErpNet.FP.Server.Services.KeepAliveHostedService[0]
      Keep Alive Background Service running...
info: ErpNet.FP.Server.Services.KeepAliveHostedService[0]
      Keep Alive Background Service done.
info: ErpNet.FP.Server.Services.KeepAliveHostedService[0]
      Keep Alive Background Service running...
info: ErpNet.FP.Server.Services.KeepAliveHostedService[0]
      Keep Alive Background Service done.
info: ErpNet.FP.Server.Services.KeepAliveHostedService[0]
      Keep Alive Background Service running...
info: ErpNet.FP.Server.Services.KeepAliveHostedService[0]
      Keep Alive Background Service done.

No problem when using com1 port. 

No problem using other software with the fiscal printer (not at the same time) so no network problem. 

this is my appsettings.json

{
  "Logging": {
    "LogLevel": {
      "Default": "Information"
    }
  },
  "ErpNet.FP": {
    "AutoDetect": false,
    "ServerId": "aR5rIm76EESRhUmxVqTLjA",
    "Printers": {
      "dt315421": {
        "Uri": "bg.dt.p.isl.com://COM1"
      },
      "dt315421_1": {
        "Uri": "bg.dt.p.isl.tcp://192.168.1.94"
      }
    }
  },
  "Kestrel": {
    "EndPoints": {
      "Http": {
        "Url": "http://0.0.0.0:8001"
      }
    },
    "Limits": {
      "MaxConcurrentConnections": 100,
      "MaxConcurrentUpgradedConnections": 100,
      "MaxRequestBodySize": 20480,
      "MaxRequestHeaderCount": 50
    }
  }
}

I hear the beep of tcp connection but the status is failed.

I am attaching wireshark capture file of the transmitted packets.

[fp800.zip](https://github.com/erpnet/ErpNet.FP/files/3306602/fp800.zip)

g-vvv, 2019-06-19:
Please provide debug.log from that session. It's in the same folder as ErpNet.FP.Server.exe.

persuader, 2019-06-19:
After removing the com1 printer just.

Starting the application, version 1.0.1006.1632...
Stopping the application.

persuader, 2019-06-20:
This is with the latest master 

Starting the application, version 1.0.1606.841...
Detecting configured printers...
Trying DT315421: bg.dt.p.isl.tcp://192.168.1.94, failed
Detecting done. Found 0 available printer(s).
Keep Alive Background Service is starting.
Keep Alive Background Service running...
Keep Alive Background Service done.
Keep Alive Background Service is stopping.
Stopping the application.

wqweto, 2019-06-20:
@persuader Това започва да заприличва на support форум, а не това e идеята на issues на проектите в github.

Има ли друг s/w с който достъпваш/печаташ на 192.168.1.94 ok? Как си обясняваш `Found 0 available printer(s)`? Аз примерно в моя LAN също не виждам принтери на 192.168.1.94 просто защото на моята машина локалния subnet е 172.17.17.0/24. . .

persuader, 2019-06-20:
В момента когато се достъпва 192.168.1.94 друг s/w не се ползва (не е пуснат няма конекции етц) иначе други софтуери има и си работят без проблем с въпросния принтер със въпросното ип.Ти такъв принтер имаш ли и тествал ли си го с ErpNet.FP. Твоя събнет какво общо има с казуса? . ErpNet.FP.server праща пакет до принтера който след това остава в режим изчакване на команда (предполагам) и трябва да се рестартира от копчето.

g-vvv, 2019-06-20:
persuader, please try the new pre-release from here:
https://github.com/erpnet/ErpNet.FP/releases/tag/v.1.0.2006.1330
If the detection is working, I will close this issue.

g-vvv, 2019-06-20:
[image]

closed 2019-06-20 by g-vvv

persuader, 2019-06-20:
[image]

Thanks.