erpnet/ErpNet.FP #46: Help about print receipt

StoyanMalchev, 2019-06-25:
What is incorrect in this request?
Thanks in advance for any help.

[debug.log](https://github.com/erpnet/ErpNet.FP/files/3324267/debug.log)

g-vvv, 2019-06-25:
Do not add double quotes before and after json opening and closing brackets.
Do not do this: "{ }". Do this { }

Do not escape quotes for fields.
Do not do this: \"text\": \"alabala\". Do this "text": "alabala"

closed 2019-06-25 by g-vvv

StoyanMalchev, 2019-06-25:
Invalid property identifier character: \\. Path '', line 2, position 1.

[debug2.log](https://github.com/erpnet/ErpNet.FP/files/3324526/debug2.log)

g-vvv, 2019-06-25:
Please do not escape double quotes with backslashes. Folow the json
protocol.

StoyanMalchev, 2019-06-25:
Thanks!
Now, status is 200.
But ok is false
error code E404 - Illegal command

[ResponceText.txt](https://github.com/erpnet/ErpNet.FP/files/3324731/ResponceText.txt)
[data.txt](https://github.com/erpnet/ErpNet.FP/files/3324732/data.txt)

i-argentinski, 2019-06-25:
- Attach debug.log to analyze this "Error occured while opening new fiscal receipt"

Also, you have another error. Your receipt should have at least one actual type=sale line. You have only comment lines, which is not ok.

Suggestion: Just remove the "isComment" field.

i-argentinski, 2019-06-25:
Also, your sale line should have amount. Also - taxGroup.

Like this:
```
		{
			"text": "По фактура 213232",
			"quantity": 1,
			"unitPrice": 36,
			"taxGroup": 2
		}
```

StoyanMalchev, 2019-06-25:
[debug3.log](https://github.com/erpnet/ErpNet.FP/files/3324847/debug3.log)
[data.txt](https://github.com/erpnet/ErpNet.FP/files/3324848/data.txt)
[ResponceText.txt](https://github.com/erpnet/ErpNet.FP/files/3324850/ResponceText.txt)

g-vvv, 2019-06-25:
See the documentation about Tremol here:

Popular Bulgarian Fiscal Printers Setup
https://github.com/erpnet/ErpNet.FP/blob/master/BGFiscalPrinterSetup.md

StoyanMalchev, 2019-06-27:
Tested on TREMOL S25.
Thanks again!

Stoyan Malchev