erpnet/ErpNet.FP #47: Retain COM port open

palex, 2019-06-29:
**ErpNet Retains COM port open although there is no printer**
After autodection phase ErpNet continue without close serial port that no printer detected.

**POS terminals, scales or other hardware on this ports will not to work**
If we have POS terminals, scales or other hardware that communicating occasionally they will fail on next try, because port is already opened.

I am not a C# master but suggest fixing in Transport.cs -> ComTransport.cs
```csharp
        public override void CloseChannel(string address)
        {
            if (openedChannels.TryGetValue(address, out Channel? channel))
            {
                channel.Dispose();
                openedChannels.Remove(address);
            }
        }
```
and in Provider.cs after drivers iteration try:
```csharp
                    if (!Found)
                    {
                        transport.CloseChannel(availableAddress.address); 
                    }
```
That fix is working for me.

closed 2019-07-02 by g-vvv

palex, 2019-07-02:
gute Arbeit

g-vvv, 2019-07-02:
Vielen Dank, mein alter Freund :)