erpnet/ErpNet.FP #54: The command is not allowed in the current fiscal mode?

StoyanMalchev, 2019-07-09:
DP-150
Режим 1: РЕГИСТРИРАНЕ
Problem on printing receipt:
[debug.log](https://github.com/erpnet/ErpNet.FP/files/3373285/debug.log)

iveivanov, 2019-07-09:
Hello, 
Try to print Z report first.

StoyanMalchev, 2019-07-09:
Same problem after Z REPORT:
[debug.log](https://github.com/erpnet/ErpNet.FP/files/3373455/debug.log)

g-vvv, 2019-07-09:
Please read the info message: "A service receipt with 90-degree rotated text printing is open".
So now, you can solve your problem with closing the previously opened 90-degree non-fiscal rotated text receipt.

closed 2019-07-09 by g-vvv

StoyanMalchev, 2019-07-10:
About  message   ...90-degree rotated text
I accept reply from DATECS:
> в модел DP150, както и във всички устройства от тяхната група, нямаме печат на бон, завъртян на 90 градуса.

iveivanov, 2019-07-10:
What happens if you try to print fiscal receipt thru the keyboard of fiscal device?

StoyanMalchev, 2019-07-10:
No problems when print thru keyboard.

g-vvv, 2019-07-10:
Please close previously opened non-fiscal receipt, no mater if it is
rotated or not.