erpnet/ErpNet.FP #56: DATECS DP-150 no longer works

dilyanrusev, 2019-07-10:
**Describe the bug**
DP-150 used to be detected and work about a month ago, but can no longer be detected after the latest version is detected.

**Describe the device(s) you are using**
DATECS DP-150

**How To Reproduce**

* Connect device to computer on Windows 10. Open device manager, and make sure that the usb port is working, and that a virtual com port is created, and that there are no issues with the device itself.
* Put the device in mode for communicating with PC.
* Start the server via `dotnet run` inside ErpNet.FP.Server

**Expected behavior**
Device is detected and can be used.
[debug.log](https://github.com/erpnet/ErpNet.FP/files/3376487/debug.log)
[image]

**Screenshots**
If applicable, add screenshots to help explain your problem.
Include debug.log if it's applicable.

**Additional context**
Add any other context about the problem here.

dilyanrusev, 2019-07-10:
After a few minutes, I noticed my laptop had turned into a vacuum cleaner. It turns out the library was consuming 100% of the CPU, even though I didn't make any requests.

g-vvv, 2019-07-10:
Please give us your appconfig here.

dilyanrusev, 2019-07-10:
Github doesn't like .json files, so I added .txt extension
[appsettings.json.txt](https://github.com/erpnet/ErpNet.FP/files/3376733/appsettings.json.txt)

g-vvv, 2019-07-10:
DP 150 is not bg.dt.p.isl, it is bg.dt.c.isl. Please change it in the
config and it will work.

dilyanrusev, 2019-07-11:
Perhaps I failed to explain the problem.
Autodetection doesn't work. I opened debug.log, and started trying out different IDs with COM3 (the port in Device Manager) via the UI. The URI that you see in the config is just the last URI that I tried.

When I use `bg.dt.c.isl://COM3`, it still doesn't work.

Here is the latest from debug.log;

```Probing bg.dt.x.isl.com://COM3... 
Request(5A): '1'
>>>1 30 30 32 3B 21 30 30 35 3A 31 5 30 31 3F 33 3 
<<<1 
<<<30 30 32 3B 21 30 30 35 3A 31 5 30 31 3F 33 3 
Probing bg.dt.c.isl.com://COM3... 
Request(5A): '1'
>>>1 25 21 5A 31 5 30 30 3D 36 3 
<<<1 
<<<25 21 5A 31 5 30 30 3D 36 3 
Probing bg.dt.p.isl.com://COM3... 
Request(5A): '1'
>>>1 25 21 5A 31 5 30 30 3D 36 3 
<<<1 
<<<25 21 5A 31 5 30 30 3D 36 3 
Probing bg.ed.isl.com://COM3... 
Request(5A): '1'
>>>1 25 21 5A 31 5 30 30 3D 36 3 
<<<1 
<<<25 21 5A 31 5 30 30 3D 36 3 
Probing bg.dy.isl.com://COM3... 
Request(5A): '1'
>>>1 25 21 5A 31 5 30 30 3D 36 3 
<<<1 
<<<25 21 5A 31 5 30 30 3D 36 3 
Probing bg.in.isl.com://COM3... 
Request(5A): '1'
>>>1 25 21 5A 31 5 30 30 3D 36 3 
<<<1 
<<<25 21 5A 31 5 30 30 3D 36 3 
Probing bg.zk.zfp.com://COM3... 
Request(60): ''
>>> Ping >>> 9 <<< 9 Request(21): ''
>>> Ping >>> 9 <<< 9 Probing bg.zk.v2.zfp.com://COM3... 
Request(60): ''
>>> Ping >>> 9 <<< 9 Request(21): ''
>>> Ping >>> 9 <<< 9 Detecting configured printers...
Trying testytest: bg.dt.c.isl://COM3, failed
Detecting done. Found 0 available printer(s).
-- HTTP Response -- Code: 200, Body(0):
-- HTTP Request -- GET: http://192.168.0.43:8001/printers, Body(0):
-- HTTP Response -- Code: 200, Body(2):
{}
-- HTTP Request -- GET: http://192.168.0.43:8001/service/printers, Body(0):
-- HTTP Response -- Code: 200, Body(59):
{"testytest":{"id":"testytest","uri":"bg.dt.c.isl://COM3"}}
-- HTTP Request -- POST: http://192.168.0.43:8001/service/printers/delete, Body(18):
{"id":"testytest"}
-- HTTP Response -- Code: 200, Body(0):
-- HTTP Request -- POST: http://192.168.0.43:8001/service/printers/configure, Body(45):
{"id":"testytest","uri":"bg.dt.c.isl://COM3"}
-- HTTP Response -- Code: 200, Body(0):
-- HTTP Request -- GET: http://192.168.0.43:8001/service/detect, Body(0):
Autodetecting local printers...
Probing bg.dt.x.isl.com://COM3... 
Request(5A): '1'
>>>1 30 30 32 3B 21 30 30 35 3A 31 5 30 31 3F 33 3 
<<<1 
<<<30 30 32 3B 21 30 30 35 3A 31 5 30 31 3F 33 3 
Probing bg.dt.c.isl.com://COM3... 
Request(5A): '1'
>>>1 25 21 5A 31 5 30 30 3D 36 3 
<<<1 
<<<25 21 5A 31 5 30 30 3D 36 3 
Probing bg.dt.p.isl.com://COM3... 
Request(5A): '1'
>>>1 25 21 5A 31 5 30 30 3D 36 3 
<<<1 
<<<25 21 5A 31 5 30 30 3D 36 3 
Probing bg.ed.isl.com://COM3... 
Request(5A): '1'
>>>1 25 21 5A 31 5 30 30 3D 36 3 
<<<1 
<<<25 21 5A 31 5 30 30 3D 36 3 
Probing bg.dy.isl.com://COM3... 
Request(5A): '1'
>>>1 25 21 5A 31 5 30 30 3D 36 3 
<<<1 
<<<25 21 5A 31 5 30 30 3D 36 3 
Probing bg.in.isl.com://COM3... 
Request(5A): '1'
>>>1 25 21 5A 31 5 30 30 3D 36 3 
<<<1 
<<<25 21 5A 31 5 30 30 3D 36 3 
Probing bg.zk.zfp.com://COM3... 
Request(60): ''
>>> Ping >>> 9 <<< 9 Request(21): ''
>>> Ping >>> 9 <<< 9 Probing bg.zk.v2.zfp.com://COM3... 
Request(60): ''
>>> Ping >>> 9 <<< 9 Request(21): ''
>>> Ping >>> 9 <<< 9 Detecting configured printers...
Trying testytest: bg.dt.c.isl://COM3, failed
Detecting done. Found 0 available printer(s).
-- HTTP Response -- Code: 200, Body(0):
-- HTTP Request -- GET: http://192.168.0.43:8001/printers, Body(0):
-- HTTP Response -- Code: 200, Body(2):
{}
-- HTTP Request -- GET: http://192.168.0.43:8001/service/printers, Body(0):
-- HTTP Response -- Code: 200, Body(59):
{"testytest":{"id":"testytest","uri":"bg.dt.c.isl://COM3"}}
```

If I delete the manually added printer, it doesn't work. A month ago this printer was printing fiscal receipts.

Also, I can't shut it down. See third picture. 100% CPU and 12gigs of memory. If it weren't so big, I'd upload a snapshot for debugging and share it.
[image]
[image]
[image]

dilyanrusev, 2019-07-11:
Note: I use the latest version from github, 

```
PS C:\dev\projects\ErpNet.FP> git log -1
commit b684a1d442be6f435087f4aa673eaf923b78a1f3 (HEAD -> master, origin/master, origin/HEAD)
Author: g-vvv <37645906+g-vvv@users.noreply.github.com>
Date:   Mon Jul 8 19:44:49 2019 +0300
```

g-vvv, 2019-07-12:
Did you follow the guidelines in https://github.com/erpnet/ErpNet.FP/blob/master/BGFiscalPrinterSetup.md ?

g-vvv, 2019-07-12:
I can not reproduce the problem in Berlin with the fiscal printer I have here.  When I come to Sofia after 8-9 days, we can try it together.

g-vvv, 2019-07-14:
When I put this to analysis:
---
Probing bg.dt.c.isl.com://COM3... 
Request(5A): '1'
>>>1 25 21 5A 31 5 30 30 3D 36 3 
<<<1 
<<<25 21 5A 31 5 30 30 3D 36 3 
--
I found that you do not follow the guidelines for Datecs in: https://github.com/erpnet/ErpNet.FP/blob/master/BGFiscalPrinterSetup.md

That means that the fiscal device will not respond to 0x5A command, as it does, when it is not in the right mode. So i will close the issue again.

closed 2019-07-14 by g-vvv

StoyanMalchev, 2019-07-15:
Problem with DP-150
[debug.log](https://github.com/erpnet/ErpNet.FP/files/3391259/debug.log)
[image]

StoyanMalchev, 2019-07-15:
Problem with DP-150
http://localhost:8001/printers/dt829313/receipt
{"type":"info","text":"A service receipt with 90-degree rotated text printing is open"}
[debug.log](https://github.com/erpnet/ErpNet.FP/files/3391380/debug.log)