erpnet/ErpNet.FP #58: Problem with reversal receipt

voltmachine, 2019-07-16:
Prints annulated receipt on this request on DP-25:

Request:
`{"uniqueSaleNumber":"DT522935-0116-0000704","receiptNumber":"0012769","receiptDateTime":"2019-07-12T17:31:02","fiscalMemorySerialNumber":"02522935","reason":"refund","items":[{"text":"\u0412\u0418\u0420\u0410\u0423\u0422 \u0442\u0430\u0431\u043b. * 20","quantity":1,"unitPrice":"29.50","taxGroup":2},{"text":"\u0412\u0418\u0420\u0410\u0423\u0422 \u0442\u0430\u0431\u043b. * 20","quantity":1,"unitPrice":"29.50","taxGroup":2},{"text":"\u0414\u043e\u0441\u0442\u0430\u0432\u043a\u0430","quantity":1,"unitPrice":4,"taxGroup":2}],"payments":[{"amount":63,"paymentType":"cash"}]}`

Response:
`{"taskStatus":"finished","result":{"ok":false,"messages":[{"type":"error","code":"E199","text":"General error"},{"type":"error","code":"E404","text":"The command is not allowed in the current fiscal mode"},{"type":"info","text":"A fiscal receipt is open"},{"type":"info","text":"SW7=OFF, SW6=OFF, SW5=OFF, SW4=OFF, SW3=OFF, SW2=OFF, SW1=OFF"},{"type":"info","text":"Unique Printer ID and Fiscal Memory ID are set"},{"type":"info","text":"BULSTAT UIC is set"},{"type":"info","text":"The tax rates are set at least once"},{"type":"info","text":"The printer is in fiscal mode"},{"type":"info","text":"The fiscal memory is formatted"},{"type":"info","text":"Error occurred in Payment 1"},{"type":"info","text":"Error occured while printing receipt items"}]}}`

[debug.log](https://github.com/erpnet/ErpNet.FP/files/3396506/debug.log)

g-vvv, 2019-07-16:
I did not found which version of ErpNet.FP you use. Please try with newest
version.

voltmachine, 2019-07-16:
The version is: 1.0.708.1935

g-vvv, 2019-07-16:
Read the Fiscal Device Manual, you cannot put paymentType: "check" in reversal receipt for that Fiscal Device. So the error is not generated by the library. The device does not give us to do the reversal receipt with "check" paymentType. I will close this issue, because is not related to the library.

closed 2019-07-16 by g-vvv

voltmachine, 2019-07-17:
But the payment is in cash:
`payments":[{"amount":63,"paymentType":"cash"}]`

g-vvv, 2019-07-17:
Then run z-report, and try again. The problem is not with the library.
Check Datecs manual.