erpnet/ErpNet.FP #59: Tax memory report from-to date/service receipt

persuader, 2019-07-18:
Datecs fiscal printers support printing of tax reports with date - for monthly and annual tax reports - command 4Fh (79) and
printing of service receipts - command 26H (38).

Can you please implement this. 

Thanks in advance.

g-vvv, 2019-07-18:
Printing of service receipts is not available by design, and never will be, because the library comply to the requirements of NRA.
Periodic reports can be done.

moynzzz, 2019-07-22:
@g-vvv What do you mean by: 

> Periodic reports can be done.

Is it possible now to make week, monthly or yearly reports or it can be implemented in the future?

g-vvv, 2019-07-22:
I'm sorry for the misunderstanding, I mean, it can be done from the service software of the cash register. 
Such reports are not the purpose of this library.

closed 2019-07-22 by g-vvv

persuader, 2019-07-23:
By law, you must make monthly report i think by the 7th day of the month and yearly report.It will be useful to implement this feature in the library as the operator must shut down the erpnet software to use the manufacturer software if the fp uses com port.