erpnet/ErpNet.FP #66: Fiscal receipt paid with "card" printed as "С ЧЕК"

moynzzz, 2019-07-25:
**Describe the bug**
When I send a request to print a receipt to my device. On the fiscal receipt for payment method it is written "С ЧЕК". 

**Describe the device(s) you are using**
The device I am using is Datex FP-2000.

**How To Reproduce**
Send a receipt request that will be paid with payment type "card".

**Expected behavior**
On the fiscal receipt it should be written "ДЕБ. КАРТА" instead of "ЧЕК".

**Screenshots**
[image]

**Request**
```json
{
    "operator": 1,
    "operatorPassword": "0000",
    "items": [
        {
            "text": "Озаряваща терапия за ръце",
            "quantity": 1,
            "unitPrice": "1.00",
            "taxGroup": 2,
            "priceModifierValue": "50.00",
            "priceModifierType": "discount-percent"
        }
    ],
    "payments": [
        {
            "paymentType": "card",
            "amount": 0.5
        }
    ]
}
```

moynzzz, 2019-07-26:
I think I found the problem:

On file `ErpNet.FP.Core/Drivers/BgDatecs/BgDatecsPIslFiscalPrinter.Commands.cs` line `130` it is stated that payment with card is with character "C" on the Datecs manual it is stated that card is with character "D".

Code of file:
```c#
switch (paymentType)
{
    case PaymentType.Cash:
        return "P";
    case PaymentType.Card:
        return "C"; // I think this is the wrong sign
    case PaymentType.Reserved1:
        return "D";
    default:
        throw new StandardizedStatusMessageException($"Payment type {paymentType} unsupported", "E406");
}
```

On Datecs manual it states the following:

> Optional code indicating the payment method. It can have the following values:
‘P’ - Cash payment (default);
‘N’ - Credit payment;
‘C’ - Check payment;
‘D’ - Debit card payment

So the solution will be to change that "C" with "D"? I can open a pull request if you want?

g-vvv, 2019-07-26:
[image]
We have another table from Datecs, related to the codes they send to NRA. So we will check before the change.

moynzzz, 2019-07-26:
I can try to build the project with "D" instead of "C" and test on the fiscal printer FP-2000 if it would be of any help?

g-vvv, 2019-07-26:
We do not have a problem changing the code, but first we need to investigate whether by submitting code "D" it will go to the NRA as a debit card payment or as NZOK payment. So, we will not change anything before consulting with Datecs.

g-vvv, 2019-07-26:
JFYI see this issue #40

moynzzz, 2019-07-26:
Thank you I will see it? Can I help to see what will be sent to NRA when the payment is "D" and also what do you think of the last statement from my previous comment "From the table you attached does that mean that we should send "NC" instead of just "C"?" or I misunderstood the table?

g-vvv, 2019-07-26:
Left column of the table shows what will be the code in the NRA XML request from the fiscal device. Right columns is the payment code, which is always single character. Communication between fiscal device and NRA, you can not listen or interfere, so you can not help us to see what will be send as XML request to NRA. For example "N" is for credit card, "C" is for debit card. But both are reported to NRA as "SCards".

iveivanov, 2019-07-26:
It seems that there are more than one variant of this protocol. I have sent a qiestion to Datecs.

moynzzz, 2019-07-26:
Okay I understood thank you for your fast responses and hard work!

g-vvv, 2019-07-26:
In the end, @moynzzz, your proposition appears to be accurate, but we should check the Datecs before applying the change. Thank you for the report.

moynzzz, 2019-07-29:
Do we have an answer from Datecs?

iveivanov, 2019-07-29:
Sorry, but we still don't have answers.

g-vvv, 2019-07-30:
Получихме следната информация:
[image]

g-vvv, 2019-07-30:
We will change the mappings for the bg.dt.p.isl according to the given table. Thank you @moynzzz , for your feedback.

closed 2019-07-31 by g-vvv