erpnet/ErpNet.FP #68: Allow to print duplicate on slip number

rosen4o, 2019-07-29:
Sometimes duplicate of slip is needed when need to be attached to invoice or warranty card
Datecs allow to print duplicate on slip from KLEN with command 7d

X protocol:  3<tab>slip_id<tab>0</tab>
old: 3,slip_id,0

i-argentinski, 2019-08-01:
Currently, we have no plans to support such feature.

But soon, there will be a request for sending raw data to the printer. See:
https://github.com/erpnet/ErpNet.FP/issues/75

You can use this to send any command to the printer.

g-vvv, 2019-12-28:
Reconsidering the implementation of printing of receipt with copy.

g-vvv, 2019-12-28:
The implementation will include new URI or header variable ("copies" as uri var) ("Copies" as header var), which will have the value "1", if the copy of the receipt is needed. For now the only possible value is "1" because most of the devices does not allow more than one copy of the receipt.

iveivanov, 2020-01-16:
It is enough to print the last receipt despite multiuser work of ErpNet.FP. The most common case is a damage of paper and nobody else can print receipt in this time.

closed 2020-01-28 by g-vvv