erpnet/ErpNet.FP #70: Payment amount difference

voltmachine, 2019-07-30:
The following receipt cannot be printed because amount difference in this partially count
item 0.625 * 8.30 = 5.1875
Items total is 5.1875, but we cannot use number 5.1875 as payment, it must be 5.19

Maybe a new OR condition can be added to issue #53 :

> receipt submitted is less than or equal to 0.001% **OR is less than 0.01**

Response:
```{  
   "taskStatus":"finished",
   "result":{  
      "ok":false,
      "messages":[  
         {  
            "type":"error",
            "code":"E403",
            "text":"Payment total amount (5.19) should be the same as the items total amount (5.18750)"
         }
      ]
   }
```

Request:
```
{  
   "uniqueSaleNumber":"DT507225-0221-0000011",
   "items":[       
      {  
         "text":"\u0424\u0415\u0420\u0412\u0415\u041a\u0421 \u0417\u0410 \u0412\u042a\u0417\u0420\u0410\u0421\u0422\u041d\u0418 \u043f\u0440\u0430\u0445 * 8",
         "quantity":0.625,
         "unitPrice":"8.30",
         "taxGroup":2
      }
   ],
   "payments":[  
      {  
         "amount":5.19,
         "paymentType":"cash"
      }
   ]
}

```

g-vvv, 2019-07-30:
This is perfect solution: "receipt submitted is less than or equal to 0.001% OR is less than 0.01".
It will be implemented. Thank you!

g-vvv, 2019-07-30:
New version of binaries in pre-release.

closed 2019-07-30 by g-vvv