erpnet/ErpNet.FP #71: Tremol fiscalMemorySerialNumber

dsetekh, 2019-07-30:
Device: Tremol FP03

Reproduce Request ( any OK status fiscal receipt request ):

> {
>   "uniqueSaleNumber": "ZK111111-0001-0000700",
>   "items": [
>     {
>       "text": "TEST_1",
>       "quantity": 1,
>       "unitPrice": 19.99,
>       "taxGroup": 2
>     }
>    ],
>   "payments": [
>       {
>       "amount": 19.99,
>       "paymentType": "cash"
>     }
>   ]
> }

Current Response:

> {
>     "receiptNumber": "000085",
>     "receiptDateTime": "2019-07-30T16:45:10",
>     "receiptAmount": 0,
>     "fiscalMemorySerialNumber": "",
>     "ok": true,
>     "messages": []
> }

Expected Response:

> {
>     "receiptNumber": "000085",
>     "receiptDateTime": "2019-07-30T16:45:10",
>     "receiptAmount": 0,
>     "fiscalMemorySerialNumber": "1234567",
>     "ok": true,
>     "messages": []
> }

! On error response the fiscalMemorySerialNumber is correctlly populated !

i-argentinski, 2019-07-30:
The fiscal printer should be fiscalized. Otherwise, some of the functions are not working with Tremol.

closed 2019-07-30 by i-argentinski

dsetekh, 2019-07-30:
Sorry didn't know that. My bad