erpnet/ErpNet.FP #72: Q: Extended Fiscal Receipt request

dsetekh, 2019-07-30:
In previous version we were able to send "amount" > Total and that way on the Receipt we had the 
"In cash" / "Change" { В брой лв / Ресто }

What is the syntax for Extended Fiscal Receipts as it is not covered in PROTOCOL.MD

> [image]

g-vvv, 2019-07-31:
In recent versions, we have rigorously verified payment amounts to ensure that the fiscal receipt is correctly printed and not misleading because of the different rounding between consumer software and cash registers.
In order to be able to compare the amount of payments with the final amount of the fiscal receipt by the library, it is necessary to know what is expected by consumer software to be received as a "change". Therefore, for payments with a payment with "change", we may need to introduce a new type of payment "type":"change" with negative amount. For your example, the payments will be two: one with "type": "cash", with amount: 50.00, and one with "type" : "change" and amount: -24.01.
We are open to any comments on this case.

dsetekh, 2019-07-31:
In ver.1.0.702.2022 we can send Request :

`{
  "uniqueSaleNumber": "ZK139294-0001-0000700",
  "items": [
    {
      "text": "аксесоари",
      "quantity": 1,
      "unitPrice": 21.66,
      "taxGroup": 2
    }
  ],
  "payments": [
      {
      "amount": 50.00,
      "paymentType": "cash"
    }
   ]
}`

This will print the Fiscal receipt from the image. So the Fiscal unit automatically handles the data and prints 'В брой' and 'Ресто' ( looks like it ). 
In the current version "amount" must equal "unitPrice" so we can not make the fiscal unit print extended receipt.
Can we mimic the behaviour from the older versions by adding a new variable let's say "sumincash" where we send the cash value?
Like:

```
...
"unitprice":12.00,
...

"payments":[

      {
      "amount": 12.00,
      "sumincash": 50.00,
      "sumchange": 38.00, // Hopefully the return change will be auto calculated by the fiscal unit so this line will not be needed
      "paymentType": "cash"
    }
]
```

This should only work for paymentType:cash as I see it

g-vvv, 2019-07-31:
```json
...
  "payments": [
    {
      "amount": 50.00,
      "paymentType": "cash"
    },
    {
      "amount": -24.01,
      "paymentType": "change"
    }
  ]
```

Тhere can be many payments, one of them can be "change". Another example:
```json
...
  "payments": [
    {
      "amount": 10.00,
      "paymentType": "card"
    },
    {
      "amount": 50.00,
      "paymentType": "cash"
    },
    {
      "amount": -34.01,
      "paymentType": "change"
    }
  ]
```

dsetekh, 2019-08-01:
Looking at most units protocol their logic is :
IF "amount" > Total [ sum(unitprice) ] then print_change; 

> If the amount received is equal to or greater than the grand total amount (the amount due) the FD quits the procedure and calculates the change in the specified type of  payment

I guess we should send what bill value the customer gives and the "amount" to be paid / that mandatory must equal the  sum("unitprice")  / and from those the "change" to be auto-calculated

Implementation sugestion :
To have 2nd command like : RECEIPTEX that takes as parameters all 3 values - The customer bill value ( "incash" ), the Total ammount of purchase ( "amount" / type:cash / ) and the change returned ( "change" / "incash" - "amount" ) { also "change" can be auto calculated based on first two } 
That way you can verify that the print is absolutely correct and there will be easy separation in terms of commands about printing simple and extended receipts.

g-vvv, 2019-08-02:
Ok, so confirmed: The variant with additional "paymentType":"change", with negative "amount" will be introduced, when there is payment with "change".

g-vvv, 2019-08-02:
Example receipt and the result:
```json
{
  "uniqueSaleNumber": "DT279013-DD01-0000001",
  "items": [
    {
      "text": "Cheese",
      "quantity": 1,
      "unitPrice": 12,
      "taxGroup": 2
    },
    {
      "type": "comment",
      "text": "Additional comment to the cheese..."
    },
    {
      "text": "Milk",
      "quantity": 2,
      "unitPrice": 10,
      "taxGroup": 2,
      "priceModifierValue": 10,
      "priceModifierType": "discount-percent"
    }
  ],
  "payments": [
    {
      "amount": 50,
      "paymentType": "cash"
    },
    {
      "amount": -20,
      "paymentType": "change"
    }
  ]
}
```
[image]

closed 2019-08-02 by g-vvv

dsetekh, 2019-08-02:
Thanks. To be tested.