erpnet/ErpNet.FP #76: Item total

voltmachine, 2019-08-01:
The following receipt with partial count items cannot be printed
0.5 * 15.05 = 7.53
0.5 * 15.05 = 7.53
Total is 15.06 and we send as payment 

But BgFiscalPrinter.cs calc **itemsTotalAmount** 15.05, because no rounding in each row and display error.

We tried to send the same receipt with payment 15.05 to pass ErpNet.FP defense. Then the receipt is annulated because, fiscal device calc price 15.06 and payment amount not fit. 

To make **itemsTotalAmount** calc like device, will need round itemPrice to second digit:

itemsTotalAmount += itemPrice;

https://github.com/erpnet/ErpNet.FP/blob/d46714f0e3265ecece4826a50beddb2b3e043b64/ErpNet.FP.Core/Drivers/BgFiscalPrinter.cs#L149

g-vvv, 2019-08-01:
There are two kinds of rounding (to the second digit). Which kind of rounding is used by the device, because we should use the same method for Midpoint rounding? 
A) Away From Zero - When a number is halfway between two others, it is rounded toward the nearest number that is away from zero.
B) To Even - When a number is halfway between two others, it is rounded toward the nearest even number.

wqweto, 2019-08-01:
Notice that `quantity` and `unitPrice` are rounded by fiscal devices to 3 and 2 digits past floating-point too.

The rounding is always A) Away From Zero. Banker's rounding B) is pretty uncommon.

g-vvv, 2019-08-01:
https://docs.microsoft.com/en-us/dotnet/api/system.midpointrounding?view=netcore-3.0
There are 3 more methods: To NegativeInfinity, To PositiveInfinity, To Zero.

g-vvv, 2019-08-01:
So, we will use the Midpoint rounding "Away from Zero"? Please confirm, and we will implement it.
Of course for quantity and unitPrice also.

voltmachine, 2019-08-01:
Yes, in current case the device use "Away from Zero" method.

g-vvv, 2019-08-01:
Ok, so confirmed. Before computation rounding of quantity (3-rd digit) and unitPrice (2-nd digit), after multiplication the result will be rounded to the second digit. Method Away from Zero.

closed 2019-08-01 by g-vvv