erpnet/ErpNet.FP #77: Protocol design: Allow comments after receipt payments

wqweto, 2019-08-02:
**Is your feature request related to a problem? Please describe.**
Current protocol design does not allow printing comment lines after receipt payments while most fiscal devices are able to print additional client info *after* totals at the bottom.

**Describe the solution you'd like**
A solution could be in `POST /printers/dt517985/receipt` to allow
```
    {
      "type": "comment",
      "text": "Additional comment to the cheese..."
    },
```
. . . under `payments` node the way it's implemented in `items`.

**Describe alternatives you've considered**
N/A

**Additional context**
N/A

g-vvv, 2019-08-02:
The preposition is this. In items array, we support two types "sale" (default), and "comment". We could support two more types - "header", "footer". 
I will give one example:
```json
{
  "uniqueSaleNumber": "DT279013-DD01-0000001",
  "operator": "1",
  "operatorPassword": "1",
  "items": [
    {
      "type": "header",
      "text": "Nice to see you again!"
    },
    {
      "text": "Cheese",
      "quantity": 1,
      "unitPrice": 12,
      "taxGroup": 2
    },
    {
      "text": "Additional comment to the cheese...",
      "type": "comment"
    },
    {
      "text": "Milk",
      "quantity": 2,
      "unitPrice": 10,
      "taxGroup": 2,
      "priceModifierValue": 10,
      "priceModifierType": "discount-percent"
    },
    {
      "type": "footer",
      "text": "Goodbye and thank you!"
    }
  ],
  "payments": [
    {
      "amount": 30,
      "paymentType": "cash"
    }
  ]
}
```
Of course the position of the "header" and "footer" items can be positioned everywhere in the items array.

g-vvv, 2019-08-02:
In future we could support type="barcode-ean-128", or type="qrcode", or type="line", etc.. but they are optional features.

g-vvv, 2019-08-02:
I have encountered a problem with this "header", "footer" thing. Header and footer can be set only in PROGRAMMING mode, which is not where we want to make changes. It is global setting, NOT Per Receipt setting, so I think, this can be done from the Fiscal Device Service Company, not by ErpNet.FP.

g-vvv, 2019-08-02:
You still can use "type": "comment" everywhere and multiple times in the items array, for writing your specific comments.
```json
{
  "uniqueSaleNumber": "DT279013-DD01-0000001",
  "operator": "1",
  "operatorPassword": "1",
  "items": [
    {
      "type": "comment",
      "text": "Nice to see you again!"
    },
    {
      "text": "Cheese",
      "quantity": 1,
      "unitPrice": 12,
      "taxGroup": 2
    },
    {
      "text": "Additional comment to the cheese...",
      "type": "comment"
    },
    {
      "text": "Milk",
      "quantity": 2,
      "unitPrice": 10,
      "taxGroup": 2,
      "priceModifierValue": 10,
      "priceModifierType": "discount-percent"
    },
    {
      "type": "comment",
      "text": "Goodbye and thank you!"
    }
  ],
  "payments": [
    {
      "amount": 30,
      "paymentType": "cash"
    }
  ]
}
```

g-vvv, 2019-08-02:
Of course, you can use new /printer/printer_id/rawrequest endpoint to send whatever you want to the fiscal device. For example, you can set up header and footer while in programming mode, for isl protocol, you can use: rawRequest = chr(149) + "P40,Nice to see you again!", rawRequest = chr(149) + "P49,Goodbye and thank you!"

wqweto, 2019-08-02:
This is about extending the protocol to allow printing *comments* like the "normal" comments in `items` but after totals section like this

[image]

Notice there is an additional comment line after both payments are printed and most devices allow this (not aware of any that does not) but it's the current protocol impl that prevents printing free-text comments once a payment is sent.

The receipt above is printed with UcsFPHub w/ this payload
```
{
    "ReceiptType": 1,
    "Operator": {
        "Code": "1",
        "Name": "Иван Иванов",
        "Password": "1"
    },
    "UniqueSaleNo": "DT518315-0001-1234567",
    "Rows": [
        {
            "ItemName": "Продукт 1",
            "Price": 12.34,
        },
        {
            "ItemName": "Продукт 2",
            "Price": 5.67,
            "TaxGroup": 2,
            "Quantity": 3.5,
            "Discount": 15
        },
        {
            "Amount": 10,
            "PaymentType": 2
        },
        {
            "PaymentType": 1
        },
        {
            "Text": "Клиентска карта: 12345"
        }
    ]
}
```

g-vvv, 2019-08-02:
Which command to use, for example in Daisy or some of Datecs protocols, to print this line, after the payments? Which command-code to use?

g-vvv, 2019-08-02:
Ok, I understand, this is regular "comment" line, but can be printed after the payments in some devices that allow this.

wqweto, 2019-08-02:
I am not aware of a device that does *not* allow printing comment lines after payments, pretty sure there is none.

If you notice on extended receipts a.k.a. invoice receipts you get from gas stations, they have all the client info (company name, address, etc.) printed *after* receipt totals as comment lines.

g-vvv, 2019-08-02:
They use "header" and "footer" global setting for printing this things. They do not use comment lines after payments.
So I do not think that we will break the structure of our protocol just for printing comment line after payments for now. But thank you for your suggestion. 
We prefer to distinguish a liability (sale items:[]) from payments:[], and not to mix them in one array Rows:[] like in your protocol. I admit that when you use an untyped array Rows:[], it gives you more flexibility, but it is loses by its readability and distinctivity. We follow different design patterns. 
If this feature is highly desired by users, we will consider some implementation in the context of our protocol.
Your comments are always helpful and appreciated.

wqweto, 2019-08-02:
> We prefer to distinguish a liability (sale items:[]) from payments:[], and not to mix them in one array Rows:[] like in your protocol. I admit that when you use an untyped array Rows:[], it gives [...]

Wait, then what happens to readability in this sample
```
    {
      "text": "Milk",
      "quantity": 2,
      "unitPrice": 10,
      "taxGroup": 2,
      "priceModifierValue": 10,
      "priceModifierType": "discount-percent"
    },
    {
      "type": "comment",
      "text": "Goodbye and thank you!"
    }
```
Why is `text` overloaded to mean both item name and comment text in different entries? It looks to me you still *are* mixing PLUs with comment lines entries in the `items` array.

I mean using `type` to disambiguate you can stuff awful lot of different entities in a god-object pattern where `text` can mean item name, comment line, payment name and operator name depending on `type` for instance.

There is nothing wrong with such `type` meta-data per se. In UcsFPHub protocol I'm disambiguating line type on sets of properties supplied and even allow passing arrays instead of JSON objects like `[ "Продукт 1", 10.50 ]` to further not repeat JSON property names and shorten http request.

Anyway. . . IMO missing ability to print footer comment lines is a short-coming of the design of the protocol as it's currently implemented. Unfortunately breaking compatibility in v2 of the API will be costly, considering the amount of development all client systems will have invested to support current version of the protocol.

Supporting `type` for "comment" with overloaded `text` in `payments` is at least backwards compatible shim.

g-vvv, 2019-08-02:
You are right. Item type="comment" and Payment type="comment" are the same, if we introduce "text" to the payment object, as it is in item object ("text"). But some of the fiscal devices supports payment with "text", so I think it can be implemented. I will reopen this issue.

wqweto, 2019-08-02:
> I will reopen this issue.

Btw, most repos leave new issues open until OP closes them or until getting referenced by a commit to the repo when github automagically closes them too (or someone finally once a month cleans abandoned ones up in one go).

wqweto, 2019-08-02:
Technically what fiscal devices support looks more like this
```json
{
   "..."
   "items":    [ 
        { "type": "item", "..." }, 
        { "type": "item", "..." }, 
        { "type": "comment", "text": "first comment" }, 
        { "type": "item", "..." }, 
        { "type": "comment", "text": "second comment" }, 
        "..." 
   ],
   "payments": [ 
        { "type": "payment", "..." }, 
        { "type": "payment", "..." }, 
        { "type": "comment", "text": "more comments" }, 
        { "type": "payment", "..." }, 
        "..." 
    ],
}
```
i.e. in `items` you can interleave PLUs and comment lines and in `payments` you can also interleave payments and comment lines.

This design effectively means that once payments are started you cannot print PLUs anymore which is usually documented in protocols payment commands.

> 35h (53) ИЗЧИСЛЯВАНЕ НА СБОР (ТОТАЛ)
> ...
> След успешното изпълнение на командата, фискалният принтер няма 
> да изпълнява командите 49 и 51 в рамките на отворения бон, обаче 
> може да изпълнява още команда 53.

After first payment you cannot add more PLUs but can print more payment and (what is not explicitly documented) more comment lines.

I think I'll settle on exactly this design with separate `items` and `payments` nodes both allowing interleaved comment lines for the UcsFPHub.

g-vvv, 2019-08-02:
Thank you, so it appears to be possible to be implemented for some of the fiscal devices. But it will be an open discussion if users really wants this feature, and will be implemented in compatible with our protocol way. The protocol is locked, we have users in production and we will not broke the compatibility for any minor post-payment comments.

g-vvv, 2019-08-02:
If something will be implemented will be for example this:
```json
{
  "uniqueSaleNumber": "DT279013-DD01-0000001",
  "operator": "1",
  "operatorPassword": "1",
  "items": [
    {
      "text": "Cheese",
      "quantity": 1,
      "unitPrice": 12,
      "taxGroup": 2
    },
    {
      "text": "Additional comment to the cheese...",
      "type": "comment"
    },
    {
      "text": "Milk",
      "quantity": 2,
      "unitPrice": 10,
      "taxGroup": 2,
      "priceModifierValue": 10,
      "priceModifierType": "discount-percent"
    },
    {
      "type": "post-payment-comment",
      "text": "Goodbye and thank you!"
    }
  ],
  "payments": [
    {
      "amount": 30,
      "paymentType": "cash"
    }
  ]
}
```

closed 2019-08-02 by g-vvv

g-vvv, 2019-12-05:
Resolved with Issue #129