erpnet/ErpNet.FP #78: Protocol design: Idempotent POST requests

wqweto, 2019-08-02:
**Is your feature request related to a problem? Please describe.**
Current protocol design does not allow idempotent POST requests. 

While `POST /printers/dt517985/receipt` always results in fiscal device state change, idempotency is important in building a fault-tolerant API. What would happen if you sent out the POST request to the server, but you get a timeout? If you repeat the request you risk duplicating the fiscal transaction.

While "Async Execution" can be used as a fault-tolerant alternative for print receipts there is no equivalent for deposts or withdrawals for instance.

**Describe the solution you'd like**
Allow some form of request identification so that repeated POST requests are *not* duplicating fiscal device state change.

**Describe alternatives you've considered**
N/A

**Additional context**
[How to achieve idempotency in POST method?](https://medium.com/@saurav200892/how-to-achieve-idempotency-in-post-method-d88d7b08fcdd)

i-argentinski, 2019-08-02:
I am writing this without testing it first, but...:

I believe, asyncTimeout is working for deposit/withdraw also. It is just not properly documented.

Try it!

wqweto, 2019-08-02:
Ooo, that would be a very good implementation if `asyncTimeout` parameter is available on all endpoints indeed.

I mean `POST http://localhost:8001/printers/zk126720/datetime?asyncTimeout=0` is an ingenious way to treat idempotency although it comes with forced asynchronicity.

g-vvv, 2019-08-02:
All endpoints in /printers API have ?asyncTimeout parameter.
Only /printers, /printers/printer_id/status, /printers/printer_id/taskinfo are without asyncTimeout, because we need them to execute in synchronous mode. For example, you can call /printers/printer_id/status, while the queue is executing something else.

g-vvv, 2019-08-02:
So I will close the issue, because the feature is already implemented. Thank you for your comments. We should update our docs and examples.

closed 2019-08-02 by g-vvv

wqweto, 2019-08-02:
JFYI, still non-async POST methods are currently unusable for any production quality system that implements any kind of resiliency while communication with this REST service.

Providing for idempotency on POSTs for any kind of transactional service (payments or fiscal memory storage as is the case) is a best practice for service's protocol design in general IMO.

g-vvv, 2019-08-02:
Our endpoints can work in async and sync mode, so the user can choose, which one to use.

wqweto, 2019-08-02:
If the issue at hand is taken care of in the protocol then it's non-obvious to me.

Can you then put some sample in the protocol description of synchronous POST with retry on timeout (e.g. WiFi connectivity issues) that prevent duplicate transactions happening in the fiscal device.

That would be most helpful for the users of the API, so to say **the** recommended idempotent POST handling by the API creators.

g-vvv, 2019-08-02:
In production they should use asyncTimeout=0 mode, and will avoid all connectivity issues.
See https://github.com/erpnet/ErpNet.FP/blob/master/PROTOCOL.md#async-execution

wqweto, 2019-08-02:
Another approach on [Idempotent Requests in Stripe API](https://stripe.com/docs/api/idempotent_requests) using `Idempotency-Key` request header

```
curl https://api.stripe.com/v1/charges \
  -u sk_test_4eC39HqLyjWDarjtT1zdp7dc: \
  -H "Idempotency-Key: 89W1jnRFLrSoyTrD" \
  -d amount=2000 \
  -d currency=usd \
  -d description="Charge for jenny.rosen@example.com" \
  -d source=tok_amex
```
. . . which uses client-side generate V4 UUIDs and is probably using the same encoding as following snippet
```C#
// serverId is RFC7515 Guid
var serverId = Convert.ToBase64String(Guid.NewGuid().ToByteArray())
    .Substring(0, 22)
```
. . . although *any* unique string is accepted for `Idempotency-Key` it seems (no need to be a UUID exactly).

g-vvv, 2019-08-02:
We already use the same generation of UUID. So we have it as taskId.

// taskId is RFC7515 Guid
            var taskId = Convert.ToBase64String(Guid.NewGuid().ToByteArray())
                .Substring(0, 22)
                .Replace("/", "_")
                .Replace("+", "-");

g-vvv, 2019-08-02:
Please read: https://github.com/erpnet/ErpNet.FP/blob/master/PROTOCOL.md#async-execution
I do not want to repeat myself.

g-vvv, 2019-08-02:
Every PrintJob before going to the queue receives taskId:
```csharp
        public string Enqueue(PrintJob printJob)
        {
            // taskId is RFC7515 Guid
            var taskId = Convert.ToBase64String(Guid.NewGuid().ToByteArray())
                .Substring(0, 22)
                .Replace("/", "_")
                .Replace("+", "-");
            Tasks[taskId] = printJob;
            TaskQueue.Enqueue(taskId);

            // Ensure that there is a running ConsumerThread, that will process the queue
            EnsureConsumer();

            return taskId;
        }
```

g-vvv, 2019-08-02:
After that with request to /printers/taskinfo with taskId parameter you can get the status or result of the task.
```csharp
        public TaskInfoResult GetTaskInfo(string taskId)
        {
            lock (taskSyncLock)
            {
                var taskInfoResult = new TaskInfoResult();
                {
                    if (Tasks.TryGetValue(taskId, out PrintJob printJob))
                    {
                        taskInfoResult.TaskStatus = printJob.TaskStatus;
                        if (printJob.Result != null)
                        {
                            taskInfoResult.Result = printJob.Result;
                        }
                    }
                }
                return taskInfoResult;
            }
        }
```

g-vvv, 2019-08-02:
We have taskId, we have fully async mode for requests with asyncTimeout=0. So if user wants, can use that mode. If user want sync mode, or async mode with timeout, it is also possible with asyncTimeout>0, or by just avoiding the asyncTimeout parameter. We give that choice to the user.  Everybody is happy, and I think to close that subject.

g-vvv, 2019-08-02:
https://github.com/erpnet/ErpNet.FP/blob/master/PROTOCOL.md#post-print-fiscal-receipt-async