erpnet/ErpNet.FP #80: Windows 32/64 bit, folder install NOGUI switch

dsetekh, 2019-08-06:
There is a service version to cover the needed functionality, but it will be a good idea to add a /nogui switch that 'hides' the notification area icon thus preventing accidental shutdown by the client.

g-vvv, 2019-08-06:
If you run the application from another application there will be no GUI. 
But we recommend to use ErpNet.FP as a service.
 /nogui switch is not planned.

closed 2019-08-06 by g-vvv

g-vvv, 2019-08-06:
There is no notification area icon anymore. We recommend to use ErpNet.FP as a service. Use Windows Services management console or command line tools to start/stop/restart the service, or use Microsoft Services API to control the Services programatically.