erpnet/ErpNet.FP #82: Problem with reversal receipt on Datecs WP-50X

StoyanMalchev, 2019-08-08:


g-vvv, 2019-08-08:
No description. Please provide description, debug.log. Follow the rules for bug report.

g-vvv, 2019-08-08:
The version, you use is not supported. Is too old. Please update to latest release from here, and try again:
https://github.com/erpnet/ErpNet.FP/releases/latest/download/ErpNet.FP.Setup.msi

closed 2019-08-08 by g-vvv

StoyanMalchev, 2019-08-16:
Another problem with current release.
Cant print receipt.
[data.txt](https://github.com/erpnet/ErpNet.FP/files/3508137/data.txt)
[Responce.txt](https://github.com/erpnet/ErpNet.FP/files/3508139/Responce.txt)

No such problem with previous version.

i-argentinski, 2019-08-16:
Hi,

You had to be using really ancient version before :)

1. You *must* have at least one item of type "sale". Otherwise, there is nothing to fiscalize and an error is returned.

2. It is preferrable that you no longer use "isComment"=true, because it is outdated and can be removed in future versions. The new syntax allows extensibility and will be the only syntax supported in the future.

Use "type"="sale" or "type"="comment"
(See docs for more info and examples)

i-argentinski, 2019-08-16:
3. You have to provide taxGroup (perhaps taxGroup=2) when fiscalizing an amount.

g-vvv, 2019-08-16:
Please always provide debug.log. Otherwise we cannot help you.