erpnet/ErpNet.FP #86: Windows Installer - missing api-ms-win-crt-runtime-l1-1-0.dll

persuader, 2019-08-09:
When using the installer the service cannot start (so the install cannot complete) due to missing 
api-ms-win-crt-runtime-l1-1-0.dll - so if KB2999226 not previously installed the install cannot finish.

The readme file states

- Everything that is needed is included in the MSI installer.

os - windows 7 home edition - not updated for more than year ago.

g-vvv, 2019-08-09:
**Thank you for the report. So the prerequisites for the application are:**

**For Windows 8.1 and earlier versions, or Windows Server 2012 R2 and earlier versions:**
Make sure that your Windows installation is up-to-date and includes KB2999226, which can be installed through Windows Update. If you don't have this update installed, you'll see an error like the following when you launch a .NET Core application: The program can't start because api-ms-win-crt-runtime-l1-1-0.dll is missing from your computer. Try reinstalling the program to fix this problem.

**For Windows 7 or Windows Server 2008 R2:**
In addition to KB2999226, make sure you also have KB2533623 installed. If you don't have this update installed, you'll see an error similar to the following when you launch a .NET Core application: The library hostfxr.dll was found, but loading it from C:\<path_to_app>\hostfxr.dll failed.

g-vvv, 2019-08-09:
Documentation updated.
Prerequisites are mentioned on README.md.

closed 2019-08-09 by g-vvv