erpnet/ErpNet.FP #88: Fiscal server stops working every other day

moynzzz, 2019-08-12:
**Describe the bug**
I'm using the fiscal server in a production environment. There is a strange issue that happens every other day which can be described as following:

The fiscal server is started each day in the morning and in the evening the computer is shut down. Every one or two days when the fiscal server is started and I open 127.0.0.1:8001 I don't see any response. The error is NO_RESPONSE. The solution I found is to delete the temporary folder that the server creates in AppData/Local/Temp. The PC I am using is on windows 7. When the folder is deleted and I start the server again everything works fine.

I think that the problem may be that this data becomes corrupted or Windows just deletes random files from this folder and the server stops working properly.

**How To Reproduce**
I'm not sure this bug can be easily reproduced.

I will wait till it happen again to make screenshots because last time there was urgent need to start the program and didn't have enough time to make them and get the log.

g-vvv, 2019-08-12:
This may be an issue for .net core 3 preview 7 single executable and trimming features. We need to research this issue on Windows 7. Thanks for your report.

g-vvv, 2019-08-12:
Maybe we should use folder, not single executable in the next releases.

moynzzz, 2019-08-12:
Can you give me a hint where should I look when it happens next so I can give additional information?

g-vvv, 2019-08-13:
Today I will make another release without "self-containing single executable, that extracts to a temporary folder" and you can try with the new release, whether the bug is reproducing again.

closed 2019-08-13 by g-vvv