erpnet/ErpNet.FP #89: Mixed payments issue when "change" is used

dsetekh, 2019-08-12:
Incorrect handling of 'cash' + 'card' + 'change' payment type.

Device : Tremol FP03

Fiscal receipt is printed with 'false' response being sent back.

Replicate :

Request :
{
  "uniqueSaleNumber": "ZK111111-0001-0000001",
  "items": [
    {
      "text": "TEST_1",
      "quantity": 1,
      "unitPrice": 15.00,
      "taxGroup": 2
    },
     {
      "text": "TEST_2",
      "quantity": 1,
      "unitPrice": 20.00,
      "taxGroup": 2
    }
   
  ],
  "payments": [
      {
      "amount": 20.00,
      
      "paymentType": "card"
    },
          {
      "amount": 30.00,
      
      "paymentType": "cash"
    },
     {
      "amount": -15.00,
      
      "paymentType": "change"
    }
      
    
  ]
}

Response :
{
    "receiptNumber": "",
    "receiptDateTime": "0001-01-01T00:00:00",
    "receiptAmount": 0,
    "fiscalMemorySerialNumber": "",
    "ok": false,
    "messages": [
        {
            "type": "error",
            "code": "E101",
            "text": "The BaseStream is only available when the port is open."
        },
        {
            "type": "info",
            "text": "Error occurred while reading last receipt QR code data"
        }
    ]
}

Requesting 'change' from 'card' payment :

Request : 
{
  "uniqueSaleNumber": "ZK111111-0001-0000002",
  "items": [
    {
      "text": "TEST_1",
      "quantity": 1,
      "unitPrice": 15.00,
      "taxGroup": 2
    },
     {
      "text": "TEST_2",
      "quantity": 1,
      "unitPrice": 20.00,
      "taxGroup": 2
    }
   
  ],
  "payments": [

          {
      "amount": 50.00,
      
      "paymentType": "card"
    },
     {
      "amount": -15.00,
      
      "paymentType": "change"
    }
  ]
}

Response :
{
    "receiptNumber": "",
    "receiptDateTime": "0001-01-01T00:00:00",
    "receiptAmount": 0,
    "fiscalMemorySerialNumber": "",
    "ok": false,
    "messages": [
        {
            "type": "error",
            "code": "E101",
            "text": "Access to the path 'COM4' is denied."
        },
        {
            "type": "info",
            "text": "Error occurred while reading last receipt QR code data"
        }
    ]
}

g-vvv, 2019-08-12:
Please provide debug.log

g-vvv, 2019-08-12:
Its looks like connection issue.

closed 2019-08-12 by g-vvv

dsetekh, 2019-08-12:
[debug.log.1.zip](https://github.com/erpnet/ErpNet.FP/files/3493009/debug.log.1.zip)

Even on false response a receipt is printed !

Yes, you're right it looks like a connection issue. Previous versions tho weren't having such issues, but this one is used as a 'service'/ msi.
OS: Windows 8.1 ( 6.3.9600 )
Fiscal Unit : Tremol FP03

Any help would be greatly appreciated.