erpnet/ErpNet.FP #90: Fiscal receipt printed and fiscal number issued, but response is false / Timeout issues

dsetekh, 2019-08-13:
Hi guys,
I have the following issue:
Once in a while a receipt is printed and it goes into the daily report, but the response is 'false' .. 

That leads to having a receipt printed twice as the response false is triggering a retry option with the main software and there is actually a receipt printed and fiscalized, that way one sale will go twice in the fiscal memory and that is pretty critical.
Looking at the debug log the 'timeout' occurs in the close fiscal receipt or at the last commands issued at protocol level, but for some reason the fiscal unit still prints the receipt and it goes into the report/memory.

The issues are mostly timeout related and in the previous versions it wasn't such an issue.

Non related
Also I noticed that sometimes the service is keeping very high CPU usage and if I start the standalone/application version in the windows a threat exhaustion/memory exhaustion is being thrown or similar, while there is enough memory available.

Replication:
I have no idea how you can replicate it without the actual device, so the option is to do it by logs alone, a good idea is to implement extended logging option at least until production versions are out.

[debug.zip](https://github.com/erpnet/ErpNet.FP/files/3497074/debug.zip)

g-vvv, 2019-08-13:
This issue is related with idleTimer (auto com port disconnect). It should be one idleTimer per com port channel, not one idleTimer per kestrel process thread. Thank you for the report. It will be fixed.

wqweto, 2019-08-13:
> That leads to having a receipt printed twice as the response false is triggering a retry option with the main software and there is actually a receipt printed and fiscalized, that way one sale will go twice in the fiscal memory and that is pretty critical.

There was a [discussion here](https://github.com/erpnet/ErpNet.FP/issues/78) about service requests idempotence in general and strategies for implementing in ErpNet.FP so that retry on fiscal transaction does *not* double sums in fiscal memory.

How many times do you retry on failure? Do you implement some kind of back off and/or circuit breaker like explained [here](https://dzone.com/articles/understanding-retry-pattern-with-exponential-back)?

Hope this "independent" feedback can convince maintainers the issue is real :-))

g-vvv, 2019-08-13:
@wqweto, @dsetekh does not use asyncTimeout=0.
We answer you many times, that if the user does not user asyncTimeout=0 and taskId, it is not good for production systems, because there is no task tracking in synchronous mode. So please, do not make me to repeat myself hundred times.

g-vvv, 2019-08-13:
Here there is another issue, that will be resolved in a mater of minutes. The problem with idleTimer.

g-vvv, 2019-08-13:
We will make asyncTimeout=0, by default in Production systems. Thank you @wqweto, you are right.

closed 2019-08-13 by g-vvv

g-vvv, 2019-08-13:
@dsetekh please try the new binaries.

wqweto, 2019-08-13:
Unfortunately `asyncTimeout=0` does *not* solve the issue completely and definately. A `POST` with `asyncTimeout=0` might never get a reply back (a transient problem with WiFi connectivity for instance) so the client app retries w/ *another* `POST` with `asyncTimeout=0` which duplicates the fiscal transaction.

Think about it -- if `asyncTimeout=0` was a solution to idempotent requests Stripe and every other payment provider would implement it instead. . .

I'll stop here and will not mention the issue anymore. Only feedback from "the trenches" might convince you at this point after v1 release.

i-argentinski, 2019-08-13:
We are considering parameter taskId, which will allow true idempotency. I will write later.

i-argentinski, 2019-08-13:
This problem will be solved by the new taskId parameter. For more information:
https://github.com/erpnet/ErpNet.FP/issues/91