erpnet/ErpNet.FP #94: Add documentation on how to run the server with HTTPS and SSL

moynzzz, 2019-08-19:
**Is your feature request related to a problem? Please describe.**
There should be a documented way on how to run the server in HTTPS environment because this is mandatory for every web communication. HTTP is not secure and the communication can be eavesdropped. I am using a server that is accessed by IP and the communication is using HTTP. I think this is not secure at all and this should be changed ASAP. Currently I can't find documented way on how to do this.

**Describe the solution you'd like**
I think currently the server can accept HTTPS and SSL communication and we can change the host to a domain and use it with SSL but I didn't have time to understand how this can be done. This should be easily documented because lazy developers or users that do not think a lot about security will be vulnerable because the connection is not encrypted and there is no easy way or a tutorial they can follow in order to make their server secure.

**Describe alternatives you've considered**
I think there is no alternative to this problem.

i-argentinski, 2019-08-19:
The server supports https. The documentation is referred by the primary readme. However, here is a link:
https://github.com/erpnet/ErpNet.FP/blob/master/Config.md#configuring-https

NOTE: The ErpNet.FP service is *not* suggested to be publicly available.

Use it only in internal localhost/LAN environment.

closed 2019-08-19 by i-argentinski

moynzzz, 2019-08-19:
How should this `Use it only in internal localhost/LAN environment.` be achieved? For example my application is a web application hosted in AWS. In order for the application to know the response it can only trust the ERP.NET FP server with a direct connection. If it is used within javascript in the users browser the application can be misguided that the fiscal printer has done its job and wrongly mark something as paid.

g-vvv, 2019-08-23:
ErpNet.FP is a service application, not IaaS, PaaS, SaaS or something like
that. So do not confuse yourself for the purpose of the library. If you
want to work in a safe environment, use additional service connection to
your AWS service, for example: build an agent application, which runs in
the client's LAN (where is the instance of ErpNet.FP), the connection (C1)
is from YOUR agent to the instance of ErpNet.FP, connection (C2) is from
YOUR agent to YOUR AWS application on http or https or tcp service port.
After that, you can use connection (C3) to YOUR service from YOUR
javascript AJAX, which internally uses the connection C2, and later C1, to
print the fiscal receipts or to check the fiscal device status, etc.. If
this scenario is too complicated for you, try something else and less
secure.