erpnet/ErpNet.FP #95: Payment check unsupported

palex, 2019-08-22:
In accordance with the documentation for payment types:
"paymentType" - one of:
"cash" - this is the default payment type if no payment type is specified
"card" - payment by debit or credit card
"check"
"packaging"
"reserved1" - often used by government regulations for specific purposes (health reimbursment, etc.)
"reserved2"

I am sending receipt:
```
{
	"uniqueSaleNumber": "DT769373-0011-0000020","operator": "1","operatorPassword": "1",
	 "items" : [{ "text":"Milk","quantity":1,"unitPrice":21.12,"taxGroup":2}],
	 "payments" : [
		{"amount": 1.12,"paymentType" : "cash"},
		{"amount": 20,"paymentType" : "check"}
	]
}
```
And result is:
```
{
	"ok":false,
	"messages":[
		{"type":"error","code":"E406","text":"Payment type Check unsupported"},	
		{"type":"info","text":"Error occured at Payment 2"}
	]
}
```

I think "check" is the nearest value for voucher. I don't know ST or SChecks should be in NRA payment schema.

iveivanov, 2019-08-22:
С think that some of Datecs models don't support "Check" payments. Especially cash registers.
Касови апарати DP-05B, DP-25, DP-150, DP-35, WP-50
[image]

palex, 2019-08-22:
Obviously it should be "I" (SOT), but what payment type have to use in ErpNet.fp?

iveivanov, 2019-08-22:
Maybe it is not so obvious if everywhere is used SChecks for "Check" payments. Admins of the project will say their point of view.
But you are right - it is not good to have a parameter which is not supported in some devices.

i-argentinski, 2019-08-22:
Currently, the payment types are under review. We will compare current and future payment types:
- Supported by the manufacturers
- Expected by NRA
- Adequate for general document exchange.

g-vvv, 2019-08-22:
It is not a bug. So I will close this issue.

closed 2019-08-22 by g-vvv