erpnet/ErpNet.FP #99: Implementing all supported payments for all fiscal devices

g-vvv, 2019-08-27:
- Implementing all supported payments for all fiscal devices.
- Returning of all supported payments for every fiscal device in /printers deviceInfo.

Our mapping definition:
    Our "paymentType" constant = NRA constant
---------------------------------------------------------
    "cash" = "SCash"
    "check" = "SChecks"
    "coupons" = "ST"
    "ext-coupons" = "SOT"
    "packaging" = "SP"
    "internal-usage" = "SSelf"
    "damage" = "SDmg"
    "card" = "SCards"
    "bank" = "SW"
    "reserve1" = "SR1"
    "reserve2" = "SR2

closed 2019-08-27 by g-vvv

g-vvv, 2019-09-13:
В брой, "cash", "SCash"
С чек, "check", "SChecks"
Талони, "coupons", "ST"
Външни талони, "ext-coupons", "SOT"
Амбалаж, "packaging", "SP"
Вътрешна употреба, "internal-usage", "SSelf"
Повреди, "damage", "SDmg"
Кредитни/дебитни карти, "card", "SCards"
Банкови трансфери, "bank", "SW"
Резерв 1/НЗОК "reserved1", "SR1"
Резерв 2 "reserved2", "SR2"

wqweto, 2019-09-15:
Name                    | Code              | XML code
----                    | ----              | ----
В брой                  | `cash`            | `SCash`
С чек                   | `check`           | `SChecks`
Талони                  | `coupons`         | `ST`
Външни талони           | `ext-coupons`     | `SOT`
Амбалаж                 | `packaging`       | `SP`
Вътрешна употреба       | `internal-usage`  | `SSelf`
Повреди                 | `damage`          | `SDmg`
Кредитни/дебитни карти  | `card`            | `SCards`
Банкови трансфери       | `bank`            | `SW`
Резерв 1/НЗОК           | `reserved1`       | `SR1`
Резерв 2                | `reserved2`       | `SR2`