erpnet/ErpNet.FP #100: Datecs Dp25 printer not found

gma123, 2019-08-31:
I have a Datecs Dp25 Printer, I just bought it, I'm not able to connect with ErpNet.FP. 
I have setup the port in the settings section and I'm using a Mini Usb cable for the connection.
On admin page the message is - Available 0 printer(s)., and if I am adding manually, is still not available. I've made debug, and the issue comes from the serialport.open().
Here is the debug file:
[debug.log](https://github.com/erpnet/ErpNet.FP/files/3562844/debug.log)

I tried to run FPrint and DUDE and I was able to connect to the printer. Should I use the RS232 cable instead of Mini Usb ? 
Here is the settings file
{
  "Logging": {
    "LogLevel": {
      "Default": "Information"
    }
  },
  "ErpNet.FP": {
    "AutoDetect": true,
    "Printers": {
      "db4200114974": {
        "Id": "db4200114974",
        "Uri": "bg.dt.c.isl.com://COM9"
      }
    }
  },
  "Kestrel": {
    "EndPoints": {
      "Http": {
        "Url": "http://0.0.0.0:8001"
      }
    },
    "Limits": {
      "MaxConcurrentConnections": 100,
      "MaxConcurrentUpgradedConnections": 100,
      "MaxRequestBodySize": 20480,
      "MaxRequestHeaderCount": 50
    }
  }
}

g-vvv, 2019-08-31:
Did you follow the https://github.com/erpnet/ErpNet.FP/blob/master/BGFiscalPrinterSetup.md

g-vvv, 2019-08-31:
Did you check the serial connection speed, it should be 115200bps for autodetect?

g-vvv, 2019-08-31:
You give me that configured uri is bg.dt.c.isl.com://COM9, but in debug.log it says: Trying DB4200114974: bg.dt.p.isl.com://COM9, failed, which means, that you misguded us. Please, be precise when you describe the issue, we do not have the time to investigate everything for you. I will close that issue and will give you one advice. Please check that COM9 is not open by another software exclusivelly and it is locked.

closed 2019-08-31 by g-vvv

g-vvv, 2019-08-31:
And please use the latest version of the library. Old versions are not supported.