erpnet/ErpNet.FP #101: Syntax error on set time on bg.dt.x.isl

valeriterziev, 2019-09-16:
"Syntax error" when try to set time on bg.dt.x.isl, Datecs DP-25X
Server version 1.0.913

debug.log: 

```
Request(3D): '16-09-19 17:24:30'
>>>1 30 30 33 3B 61 30 30 33 3D 31 36 2D 30 39 2D 31 39 20 31 37 3A 32 34 3A 33 30 5 30 35 35 3D 3 
<<<16 
<<<1 30 30 33 3B 61 30 30 33 3D 2D 31 31 32 30 30 31 9 4 A1 80 80 80 84 82 80 80 5 30 37 38 3A 3 
Response(8): -112001	
Request(3E): ''
>>>1 30 30 32 3A 62 30 30 33 3E 5 30 32 30 34 3 
<<<16 
<<<1 30 30 34 3B 62 30 30 33 3E 30 9 31 36 2D 30 39 2D 31 39 20 31 37 3A 32 36 3A 34 34 20 44 53 54 9 4 80 80 80 80 84 82 80 80 5 30 3A 3B 3E 3 
Response(24): 0	16-09-19 17:26:44 DST	
Idle timer elapsed for the com port COM4
Closing the com port COM4
Request(3D): '16-09-19 17:24:31'
>>>1 30 30 33 3B 63 30 30 33 3D 31 36 2D 30 39 2D 31 39 20 31 37 3A 32 34 3A 33 31 5 30 35 36 30 3 
Opening the com port COM4
Request(3E): ''
>>>1 30 30 32 3A 64 30 30 33 3E 5 30 32 30 36 3 
Opening the com port COM4
Keep Alive Background Service running...
Request(3E): ''
>>>1 30 30 32 3A 65 30 30 33 3E 5 30 32 30 37 3 
Opening the com port COM4
<<<16 
<<<1 30 30 34 3B 65 30 30 33 3E 30 9 31 36 2D 30 39 2D 31 39 20 31 37 3A 32 38 3A 32 35 20 44 53 54 9 4 80 80 80 80 84 82 80 80 5 30 3A 3C 32 3 
Response(24): 0	16-09-19 17:28:25 DST	
Keep Alive Background Service done.
Idle timer elapsed for the com port COM4
Closing the com port COM4
Keep Alive Background Service running...
Request(3E): ''
>>>1 30 30 32 3A 66 30 30 33 3E 5 30 32 30 38 3 
Opening the com port COM4
<<<16 
<<<1 30 30 34 3B 66 30 30 33 3E 30 9 31 36 2D 30 39 2D 31 39 20 31 37 3A 33 30 3A 32 35 20 44 53 54 9 4 80 80 80 80 84 82 80 80 5 30 3A 3B 3C 3 
Response(24): 0	16-09-19 17:30:25 DST	
Keep Alive Background Service done.
Idle timer elapsed for the com port COM4
Closing the com port COM4
```

g-vvv, 2019-09-16:
As I see, you set the date and time multiple times, and succeded. I want to get the http request/response log but I'll need to fix the debug.log default verbocity level for ErpNet.FP. After that, with the new version you will try again.

wqweto, 2019-09-16:
IMO on this line
```
Request(3D): '16-09-19 17:24:30'
```
. . .  the params need a final `<SEP>` (`'\t'`) at the end as documented in the protocol.

```
Command: 61 (3Dh) Set date and time
Parameters of the command:
{DateTime}<SEP>
```

g-vvv, 2019-09-16:
Tomorrow I will fix the log verbocity and will override SetDateTime, with the variant with ending "\t" as mentioned in the X protocol.
Thank you for your feedback and support.

g-vvv, 2019-09-17:
Solved with the new v.1.0.917.1500

closed 2019-09-17 by g-vvv