erpnet/ErpNet.FP #106: ADDING A FISCAL DEVICE ELECTRONICA12 SVID.834/05.07.19

TSEKA12, 2019-10-02:
Здравейте,
Ние сме производител на фискално устройство. Какво трябва да направим
за да го включим във Вашия драйвер?
С Уважение: Пламен Попов

i-argentinski, 2019-10-02:
Здравейте,

Трябва някак да се свържем. 

Един вариант е във Фейсбук групата:
https://www.facebook.com/groups/BgBusinessDev/

Друг вариант е да дадете имейл.

TSEKA12, 2019-10-03:
Здравейте,
Благодарим за бързия отговор!
Моят имейл е ts@electronica.bg . Ние се намираме в завод ЕЛЕКТРОНИКА София.
Касовият ни апарат поддържа ISL протоколите на Дейзи и Датекс (поне основните и голяма
част от останалите).
Опознавателната ни команда 0x5a изпраща следния отговор:
 Nyla-1.01BG 20-08-2011 19:49,7EC1,00,6,EEnnnnnn,ffffffff
nnnnnn -номер FDIN
ffffffff-номер фискална памет
Узаконената контр.сума е 7EC1, но ни предстои преузаконяване до края на месеца,
Ако потрябва можем да изпратим протокола на комуникация, но си мисля , че е най-лесно
просто да приложите същия като на Дейзи или ако се наложи на Датекс, със горепосоченото
опознаване.
С Уважение: Пламен Попов

g-vvv, 2019-10-03:
Здравейте,

Изпратете протокола за комуникация, за да видим и съобщенията за грешка при различните статус битове и други специфични особености. Да правим имплементация без описание на протокола ще е несериозно.

Може да го приложите направо тук.

TSEKA12, 2019-10-04:
Здравейте,
Прилагам протокола:
[CPBUL_W.TXT](https://github.com/erpnet/ErpNet.FP/files/3689998/CPBUL_W.TXT).
както и инструкцията по експлоатация:
[e12iefinTN.pdf](https://github.com/erpnet/ErpNet.FP/files/3690027/e12iefinTN.pdf)
Ако желаете може да Ви изпратим касов апарат, регистриран на тестовия
сървър на НАП или Вие да ни дадете линк към някакъв Ваш тестови софтуер.
С Уважение: Пламен Попов

TSEKA12, 2019-10-07:
Здравейте,
Ще изпратим касов апарат, но бихме искали предварително да уточним някои неща:
1.ЕЛЕКТРОНИКА12 работи както на акумулаторни батерии (8 часа в режим на печат), така
и с адаптер, който ще приложим. Ще приложим и интерфейсен кабел за RS232  с CANON9.
Скоростта по дефолт ни е 9600 и предпочитаме КА да работи с нея. Ако това не е възможно
можем да Ви я установим на 115200 или да Ви кажем как можете да я променяте.
Желаете ли да я направим 115200?
2.На всички производители както знаете предстои да внесат за преузаконяване до 31.10.19
с цел клиентите да сменят фърмуерите си до 31.03.20. Протокола за комуникация , който съм 
приложил не е 100% точен и разбираем, както и ако забележите някои несъответствия
можем да изменим него или фърмуера. 
Да Ви сложим ли ВПН SIM карта за да можем дистанционно да сменяме фърмуера?
3. По принцип целта ни е протокола за комуникация да ни е максимално близък до този
на Дейзи, защото продаваме и техни КА.
 С Уважение: Пламен Попов

g-vvv, 2019-10-07:
Здравейте, благодарим за съдействието. Всички производители минаха на 115200 по подразбиране. Би било добре да минете и вие. Покажете ни как да я сменим, предполагам го има в протокола или става с джъмпери? Сложете такава карта да, сменяйте фърмуера с този, който ще е актуален. Разбира се, хубаво би било да разбираме какви са промените, за да ги отразим правилно.

TSEKA12, 2019-10-07:
Здравейте,
Днес не остана време да изпратиме касата, но утре ще я изпратим установена на 115200.
За съжаление не може да го поставим по подразбиране, но в протокола и инструкцията по
експлоатация,които съм приложил, смяната на Параметър 19 със стойност 10(115200) вместо с 0(9600) е описана.
Има и начин това да става през PC но всички тези начини ще Ви ги изпратим допълнително
описани в документ. По подразбиране паролата на оператор 1 е 1, но може да работи едновременно и без пароли (парола 0 за оператори от 1 до 10).
Да ще Ви държим в течение на всички неща.
С Уважение: Пламен Попов

TSEKA12, 2019-10-08:
Здравейте,
Изпращаме днес по куриер касата.
Имайте предвид, че може да се получи ситуация като днешната в 10:00 сървъра на НАП да не работи.
14 часа след такова нещо на индикация започва обратно броене на оставащите часове до
блокировка които са още 10. При такава ситуация касата отговаря с грешка ФП и още
някои.Някои от информационните команди се изпълняват нормално. Ако се появи връзката
касата до няколко десетки минути би трябвало автоматично да се разблокира.
Прилагам леко променен Протокол за комуникация:
[CPBUL_W.TXT](https://github.com/erpnet/ErpNet.FP/files/3701857/CPBUL_W.TXT)

С Уважение: Пламен Попов

g-vvv, 2019-10-08:
Здравейте,
Благодаря за Протокола за комуникация. Очакваме касата и ще започнем да я правим.

TSEKA12, 2019-10-11:
Здравейте,
Изпратили сме касата. Молим ако има някакви проблеми да ни съобщите за да вземем 
необходимите мерки до края на месеца, когато трябва да я подадем за функционални
изпитания на софтуерна версия.
Добре би било като я включите за първи път да я оставите за известно време. Тя може
да се самоизключи след обновяване на фърмуера и после е готова за използване.
Приятна събота и неделя.Благодаря предварително:
Пламен Попов

TSEKA12, 2019-12-20:
Здравейте Г-н Аржентински,
Нямаме отговор от Вас относно тестовите изпитания на ЕКАФП Електроника 12 с ФП 0113013,
работещ във фискален режим. Същият Ви бе изпратен по куриер на 8.10.19г. Последно имаме
регистрирани събития с него на дата 14.10.19г.
Имаме желание да включите ЕЛЕКТРОНИКА12 в списъка с касови апарати, работещи с Вашия
драйвер. Към момента ЕЛЕКТРОНИКА12 не е включена в него.
В случай, че имате  забележки, относно работата на апарата, може да ги обсъдим.
При липса на интерес от Ваша страна настояваме да върнете споменатия касов апарат на
адрес: гр.София,бул.Шипченски проход 63, завод ЕЛЕКТРОНИКА за ЕЛЕКТРОНИКА ЕАД
до края на 2019г.
Поздрави: Пламен Попов
гр.София 20.12.2019

g-vvv, 2019-12-20:
Моля, комуникирайте по имейл. Тук не е мястото за уговаряне на срещи и отношения.

closed 2019-12-20 by g-vvv