erpnet/ErpNet.FP #110: An assembly specified in the application dependencies manifest (ErpNet.FP.Server.deps.json) was not found

f1mp3r, 2019-10-02:
I am getting the following error when trying to run the project (the linux 64 build) on a Debian VM:
```
Error:
  An assembly specified in the application dependencies manifest (ErpNet.FP.Server.deps.json) was not found:
    package: 'runtimepack.Microsoft.NETCore.App.Runtime.linux-x64', version: '3.0.0'
    path: 'createdump'
```

I have installed both .NET Core 3.0 SDK and .NET Core 3.0 Runtime on the machine.
When I run `dotnet --info` on the machine - this is what I get:
```
.NET Core SDK (reflecting any global.json):
 Version:   3.0.100
 Commit:    04339c3a26

Runtime Environment:
 OS Name:     debian
 OS Version:  9
 OS Platform: Linux
 RID:         debian.9-x64
 Base Path:   /usr/share/dotnet/sdk/3.0.100/

Host (useful for support):
  Version: 3.0.0
  Commit:  95a0a61858

.NET Core SDKs installed:
  3.0.100 [/usr/share/dotnet/sdk]

.NET Core runtimes installed:
  Microsoft.AspNetCore.App 3.0.0 [/usr/share/dotnet/shared/Microsoft.AspNetCore.App]
  Microsoft.NETCore.App 3.0.0 [/usr/share/dotnet/shared/Microsoft.NETCore.App]

To install additional .NET Core runtimes or SDKs:
  https://aka.ms/dotnet-download
```

g-vvv, 2019-10-03:
https://go.microsoft.com/fwlink/?linkid=798306

g-vvv, 2019-10-03:
It is not related to the ErpNet.FP library, It is related to. net core framework, so I will close thos issue.

closed 2019-10-03 by g-vvv