erpnet/ErpNet.FP #112: Cut the comment text on DP-150

iveivanov, 2019-10-03:
**Describe the bug**
The working area for comment text is 40 symbols:
{
    "uri": "bg.dt.c.isl.com://COM17",
    "serialNumber": "DT744394",
    "fiscalMemorySerialNumber": "02744394",
    "manufacturer": "Datecs",
    "model": "DP-150",
    "firmwareVersion": "265403 08Nov18 1050",
    "itemTextMaxLength": 22,
    *"commentTextMaxLength": 40,*
    "operatorPasswordMaxLength": 8,
    "taxIdentificationNumber": "000713391",
    "supportedPaymentTypes": [
        "cash",
        "coupons",
        "ext-coupons",
        "card",
        "reserved1"
    ]
}

When I'm printing 40 symbols comment it is printed only 36:

"uniqueSaleNumber": "DY423793-0001-0000002",
	"items" : 
	[
		{
			"text": "Cheese",
			"quantity": 1,
			"unitPrice": 12,
			"taxGroup": 2
		},
		{
			"text": "AbcdefghAbcdefghAbcdefghAbcdEfghAbcdEfgh",
			"type": "comment"
		},
		{
			"text": "Milk",
			"quantity": 2,
			"unitPrice": 10,
			"taxGroup": 2,
			"priceModifierValue": 10,
			"priceModifierType": "discount-percent"
		}
	],
	"Payments" : [
		{
			"paymentType": "cash",
			"amount": "20"
		},
		{
			"paymentType": "card",
			"amount": "10"
		}
	]
}

[image]

**Describe the device(s) you are using**
Datecs DP-150

**How To Reproduce**
Print the attached JSON

**Expected behavior**
I expect to be printed all 40 symbols in the comment text

iveivanov, 2019-10-03:
Debug Log:

[2019-10-03 18:23:57 INF] bg.dt.c.isl.com://COM17 <<< 01-24-24-3C-05-30-30-38-39-03
[2019-10-03 18:23:57 INF] Opening the com port COM17
[2019-10-03 18:23:57 INF] bg.dt.c.isl.com://COM17 >>> 16
[2019-10-03 18:23:57 INF] bg.dt.c.isl.com://COM17 >>> 01-2B-24-3C-04-A0-82-80-80-86-9A-05-30-33-3D-36-03
[2019-10-03 18:23:57 INF] bg.dt.c.isl.com://COM17 <<< 01-3F-25-30-31-2C-31-2C-44-59-34-32-33-37-39-33-2D-30-30-30-31-2D-30-30-30-30-30-30-32-2C-31-05-30-35-3F-36-03
[2019-10-03 18:23:57 INF] bg.dt.c.isl.com://COM17 >>> 16
[2019-10-03 18:23:57 INF] bg.dt.c.isl.com://COM17 >>> 16
[2019-10-03 18:23:57 INF] bg.dt.c.isl.com://COM17 >>> 16
[2019-10-03 18:23:57 INF] bg.dt.c.isl.com://COM17 >>> 16
[2019-10-03 18:23:57 INF] bg.dt.c.isl.com://COM17 >>> 16
[2019-10-03 18:23:57 INF] bg.dt.c.isl.com://COM17 >>> 16
[2019-10-03 18:23:57 INF] bg.dt.c.isl.com://COM17 >>> 16
[2019-10-03 18:23:57 INF] bg.dt.c.isl.com://COM17 >>> 16
[2019-10-03 18:23:57 INF] bg.dt.c.isl.com://COM17 >>> 16
[2019-10-03 18:23:57 INF] bg.dt.c.isl.com://COM17 >>> 16
[2019-10-03 18:23:57 INF] bg.dt.c.isl.com://COM17 >>> 16
[2019-10-03 18:23:57 INF] bg.dt.c.isl.com://COM17 >>> 16
[2019-10-03 18:23:57 INF] bg.dt.c.isl.com://COM17 >>> 16
[2019-10-03 18:23:57 INF] bg.dt.c.isl.com://COM17 >>> 16
[2019-10-03 18:23:57 INF] bg.dt.c.isl.com://COM17 >>> 01-34-25-30-30-30-30-36-2C-30-30-30-36-04-80-80-88-80-86-9A-05-30-35-37-32-03
[2019-10-03 18:23:57 INF] bg.dt.c.isl.com://COM17 <<< 01-25-26-5A-31-05-30-30-3D-3B-03
[2019-10-03 18:23:57 INF] bg.dt.c.isl.com://COM17 >>> 16
[2019-10-03 18:23:57 INF] bg.dt.c.isl.com://COM17 >>> 16
[2019-10-03 18:23:57 INF] bg.dt.c.isl.com://COM17 >>> 16
[2019-10-03 18:23:58 INF] bg.dt.c.isl.com://COM17 >>> 16
[2019-10-03 18:23:58 INF] bg.dt.c.isl.com://COM17 >>> 16
[2019-10-03 18:23:58 INF] bg.dt.c.isl.com://COM17 >>> 16
[2019-10-03 18:23:58 INF] bg.dt.c.isl.com://COM17 >>> 16
[2019-10-03 18:23:58 INF] bg.dt.c.isl.com://COM17 >>> 16
[2019-10-03 18:23:58 INF] bg.dt.c.isl.com://COM17 >>> 16
[2019-10-03 18:23:58 INF] bg.dt.c.isl.com://COM17 >>> 16
[2019-10-03 18:23:58 INF] bg.dt.c.isl.com://COM17 >>> 16
[2019-10-03 18:23:58 INF] bg.dt.c.isl.com://COM17 >>> 16
[2019-10-03 18:23:58 INF] bg.dt.c.isl.com://COM17 >>> 01
[2019-10-03 18:23:58 INF] bg.dt.c.isl.com://COM17 >>> 65-26-5A-44-50-2D-31-35-30-2C-32-36-35-34-30-33-20-30-38-4E-6F-76-31-38-20-31-30-35-30-2C-35-34-32-38-2C-30-30-30-30-30-30-30-30-2C-44-54-37-34-34-33-39-34-2C-30-32-37-34-34-33-39-34-04-80-80-88-80-86-9A-05-31-30-35-32-03
[2019-10-03 18:23:58 INF] bg.dt.c.isl.com://COM17 <<< 01-33-27-31-43-68-65-65-73-65-09-C1-31-32-2E-30-30-2A-31-05-30-34-3F-33-03
[2019-10-03 18:23:58 INF] bg.dt.c.isl.com://COM17 >>> 16
[2019-10-03 18:23:58 INF] bg.dt.c.isl.com://COM17 >>> 16
[2019-10-03 18:23:58 INF] bg.dt.c.isl.com://COM17 >>> 16
[2019-10-03 18:23:58 INF] bg.dt.c.isl.com://COM17 >>> 16
[2019-10-03 18:23:58 INF] bg.dt.c.isl.com://COM17 >>> 16
[2019-10-03 18:23:58 INF] bg.dt.c.isl.com://COM17 >>> 01-2B-27-31-04-80-80-88-80-86-9A-05-30-33-3B-34-03
[2019-10-03 18:23:58 INF] bg.dt.c.isl.com://COM17 <<< 01-4C-28-36-41-62-63-64-65-66-67-68-41-62-63-64-65-66-67-68-41-62-63-64-65-66-67-68-41-62-63-64-45-66-67-68-41-62-63-64-45-66-67-68-05-30-3F-38-33-03
[2019-10-03 18:23:58 INF] bg.dt.c.isl.com://COM17 >>> 16
[2019-10-03 18:23:58 INF] bg.dt.c.isl.com://COM17 >>> 16
[2019-10-03 18:23:58 INF] bg.dt.c.isl.com://COM17 >>> 16
[2019-10-03 18:23:58 INF] bg.dt.c.isl.com://COM17 >>> 01
[2019-10-03 18:23:58 INF] bg.dt.c.isl.com://COM17 >>> 2B-28-36-04-80-80-88-80-86-9A-05-30-33-3B-3A-03
[2019-10-03 18:23:58 INF] bg.dt.c.isl.com://COM17 <<< 01-38-29-31-4D-69-6C-6B-09-C1-31-30-2E-30-30-2A-32-2C-2D-31-30-2E-30-30-05-30-35-38-31-03
[2019-10-03 18:23:58 INF] bg.dt.c.isl.com://COM17 >>> 16
[2019-10-03 18:23:58 INF] bg.dt.c.isl.com://COM17 >>> 16
[2019-10-03 18:23:58 INF] bg.dt.c.isl.com://COM17 >>> 16
[2019-10-03 18:23:58 INF] bg.dt.c.isl.com://COM17 >>> 16
[2019-10-03 18:23:58 INF] bg.dt.c.isl.com://COM17 >>> 16
[2019-10-03 18:23:58 INF] bg.dt.c.isl.com://COM17 >>> 16
[2019-10-03 18:23:58 INF] bg.dt.c.isl.com://COM17 >>> 16
[2019-10-03 18:23:58 INF] bg.dt.c.isl.com://COM17 >>> 16
[2019-10-03 18:23:58 INF] bg.dt.c.isl.com://COM17 >>> 16
[2019-10-03 18:23:59 INF] bg.dt.c.isl.com://COM17 >>> 16
[2019-10-03 18:23:59 INF] bg.dt.c.isl.com://COM17 >>> 01
[2019-10-03 18:23:59 INF] bg.dt.c.isl.com://COM17 >>> 2B-29-31-04-80-80-88-80-86-9A-05-30-33-3B-36-03
[2019-10-03 18:23:59 INF] bg.dt.c.isl.com://COM17 <<< 01-2B-2A-35-09-50-32-30-2E-30-30-05-30-31-3D-38-03
[2019-10-03 18:23:59 INF] bg.dt.c.isl.com://COM17 >>> 16
[2019-10-03 18:23:59 INF] bg.dt.c.isl.com://COM17 >>> 16
[2019-10-03 18:23:59 INF] bg.dt.c.isl.com://COM17 >>> 16
[2019-10-03 18:23:59 INF] bg.dt.c.isl.com://COM17 >>> 16
[2019-10-03 18:23:59 INF] bg.dt.c.isl.com://COM17 >>> 16
[2019-10-03 18:23:59 INF] bg.dt.c.isl.com://COM17 >>> 16
[... 62 lines cut ...]
[2019-10-03 18:24:01 INF] bg.dt.c.isl.com://COM17 >>> 16
[2019-10-03 18:24:01 INF] bg.dt.c.isl.com://COM17 >>> 16
[2019-10-03 18:24:01 INF] bg.dt.c.isl.com://COM17 >>> 16
[2019-10-03 18:24:01 INF] bg.dt.c.isl.com://COM17 >>> 16
[2019-10-03 18:24:01 INF] bg.dt.c.isl.com://COM17 >>> 16
[2019-10-03 18:24:01 INF] bg.dt.c.isl.com://COM17 >>> 16
[2019-10-03 18:24:01 INF] bg.dt.c.isl.com://COM17 >>> 16
[2019-10-03 18:24:01 INF] bg.dt.c.isl.com://COM17 >>> 16
[2019-10-03 18:24:01 INF] bg.dt.c.isl.com://COM17 >>> 16
[2019-10-03 18:24:01 INF] bg.dt.c.isl.com://COM17 >>> 16
[... cut ...]
[2019-10-03 18:24:02 INF] bg.dt.c.isl.com://COM17 >>> 9A
[2019-10-03 18:24:02 INF] bg.dt.c.isl.com://COM17 >>> 05-30-35-37-3D-03
[2019-10-03 18:24:02 INF] bg.dt.c.isl.com://COM17 <<< 01-24-2F-71-05-30-30-3C-39-03
[2019-10-03 18:24:02 INF] bg.dt.c.isl.com://COM17 >>> 16
[2019-10-03 18:24:02 INF] bg.dt.c.isl.com://COM17 >>> 01-32-2F-71-30-30-30-30-31-30-38-04-80-80-80-80-86-9A-05-30-35-35-34-03
[2019-10-03 18:24:03 INF] Idle timer elapsed for the com port COM17
[2019-10-03 18:24:03 INF] Closing the com port COM17

g-vvv, 2019-10-07:
As we can se in the data: 01-4C-28-36-41-62-63-64-65-66-67-68-41-62-63-64-65-66-67-68-41-62-63-64-65-66-67-68-41-62-63-64-45-66-67-68-41-62-63-64-45-66-67-68-05-30-3F-38-33-03
41-62-63-64-65-66-67-68 is repeated times, which means, that the library sent correct number of bytes  (40) to the COM port, so the problem is not in ErpNet.FP. Please ask Datecs about that limitation.
This is not bug in the ErpNet.FP, so I will close this issue.

closed 2019-10-07 by g-vvv