erpnet/ErpNet.FP #113: JFYI comment on issue #112

wqweto, 2019-10-07:
[Line 59 in `BgDatecsCIslFiscalPrinterDriver.cs`](https://github.com/erpnet/ErpNet.FP/blob/master/ErpNet.FP.Core/Drivers/BgDatecs/BgDatecsCIslFiscalPrinterDriver.cs#L59) is probably incorrect.
```
    CommentTextMaxLength = 40, // Set by Datecs protocol
```
The underneath calculation of 42 symbols (from 57 mm paper width) minus 2 symbols for trailing # in comment lines is not holding true in reality and 36 is a the real limit on Datecs cash-registers for unknown reasons.

I can confirm that my (recent) DP-25 model here has the same artificial 36 symbols limitation on comment lines and this is not a bug in the firmware of OP's particular model.

(Please don't close OP github issues so fast, if you want collaborators input that is.)

g-vvv, 2019-10-07:
Ok, but in reality is a firmware limitation, not a bug in ErpNet.FP..

g-vvv, 2019-10-07:
When somebody expects to write a comment spreaded between # and # it will be imposible to fill it up to. the end, which is the expectation. If I put a 36 symbols, they will. report a Bug, because it is less than 40. So the problem is within Datecs firmware.

g-vvv, 2019-10-07:
They should fix this issue on their own, but you are right, I should put 36 symbols, because otherwise the information will be cutout. So I will fix this "bug", which is defacto workaround over Datecs limitations, and the line will be not filled to the end, but there will be no cutouts.

g-vvv, 2019-10-07:
Ok, so now I will close this "bug" issue.

closed 2019-10-07 by g-vvv