erpnet/ErpNet.FP #115: Datecs FP-650 `DeviceInfo`

wqweto, 2019-10-10:
JFYI, in `BgDatecsPIslFiscalPrinterDriver.cs` the `DeviceInfo`'s members [`CommentTextMaxLength` and `ItemTextMaxLength`](https://github.com/erpnet/ErpNet.FP/blob/master/ErpNet.FP.Core/Drivers/BgDatecs/BgDatecsPIslFiscalPrinterDriver.cs#L60-L61) are off by 2 for FP-650 and probably the rest of the (lagacy) FP line.

[FP-650 product description](http://www.datecs.bg/bg/products/304) says `Име на артикул - До 36 символа` but anything above 34 bombs with a syntax error while the comment is silently trimmed at 40 symbols.

g-vvv, 2019-10-10:
So there is again one of Datecs "hidden" features :) Thanks, I will fix it.

g-vvv, 2019-10-10:
I will try with the FP-2000, if the issue apears there also.

g-vvv, 2019-10-10:
New binaries published

closed 2019-10-10 by g-vvv