erpnet/ErpNet.FP #116: Return / Exchange error thrown ( negative amount value in first position JSON )

dsetekh, 2019-10-14:
Hi guys,
the situation occurs if we have an item return and exchange.
Real world example : You get a L size shirt, but you need to exchange for XXL size shirt.
The operator does not first Storno the old receipt, but issue an 'return/exchange' receipt where one item is negative ( returned ) and one positive ( exchanged ). In the protocols ( Datecs, Tremol etc... ) it is the negative value in the sell article commands.

The JSON will be :
```
{
  "uniqueSaleNumber": "DT000000-0001-0000001",

  "items": [
    {
      "text": "Shirt L",
      "quantity": 1,
      "unitPrice": -10,
      "taxGroup": 2
    },
  
    {
      "text": "Shirt XXL",
      "quantity": 1,
      "unitPrice": 10,
      "taxGroup": 2,
    }
  ],
  "payments": []
}
```
When issued with equal amount values than 'Payment' must be empty or it will fail with 'positive value needed' ( This is another thing as sometimes we may have a 0 amount value as a total, but it can be 'mitigated' by not sending the payments at all and leave them to be auto-calculated  ).. 

The issue is when we have the negative value in first position, then we'll get an error :
```
{
    "receiptNumber": "",
    "receiptDateTime": "0001-01-01T00:00:00",
    "receiptAmount": 0,
    "fiscalMemorySerialNumber": "02830967",
    "ok": false,
    "messages": [
        {
            "type": "error",
            "code": "E199",
            "text": "General error"
        },
        {
            "type": "error",
            "code": "E404",
            "text": "The command is not allowed in the current fiscal mode"
        }
...
```
Test unit : Datecs DP-150
Test version : v.1.0.1010.1440

Thank you.
[debug.log](https://github.com/erpnet/ErpNet.FP/files/3724760/debug.log)

g-vvv, 2019-10-14:
Unit price cannot be negative. Your version of ErpNet.FP is way to old and accepts negative unitPrice. In newer versions we dissallow this, because is not supported by all fiscal devices.

g-vvv, 2019-10-14:
The way to do this is to make an reversalReceipt for the first unit (reason: refund) and create new receipt for the second unit.

closed 2019-10-14 by g-vvv