erpnet/ErpNet.FP #120: Allow empty quantity

iveivanov, 2019-11-08:
It is a good idea to allow a request without quantity in Json body.
The goal is to print receipt in this form:

[image]

We put the quantity in comment text and only text and amount in a fiscal row.

g-vvv, 2019-11-08:
[image]

g-vvv, 2019-11-08:
It not always works, when there is no quantity present. For example, for Datecs works, for Tremol, do not work, it always prints the quantity (default = 1). I will try with Daisy, Eltrade, and others.

g-vvv, 2019-11-08:
[image]

g-vvv, 2019-11-08:
Daisy, Eltrade, works!

g-vvv, 2019-11-08:
This is the sample receipt I use for these tests:
```
{  
   "uniqueSaleNumber":"DT122222-0001-0000900",
   "items":[  
      {  
         "text":"Toy",
         "unitPrice":15.00,
         "taxGroup":2
      },
      {  
         "text":"Ball",
         "unitPrice":20.00,
         "taxGroup":2
      }
   ],
   "payments":[  
      {  
         "amount":35.00,
         "paymentType":"cash"
      },
   ]
}
```

g-vvv, 2019-11-08:
Binaries for the new release: 1.0.1108.1800

closed 2019-11-08 by g-vvv