erpnet/ErpNet.FP #121: Error on printReceipt

valeriterziev, 2019-11-11:
An error received while print a specific receipt. Version 1.0.1108.

Request
```
{ 
   "uniqueSaleNumber":"DT522935-0083-0018856",
   "operator":"1",
   "operatorPassword":"1",
   "items":[ 
      { 
         "text":"\u0421\u0423\u041e\u041d\u0421\u042a\u041d \u041b\u0418\u0422\u0418\u0415\u0412 \u041e\u0420\u041e\u0422\u0410\u0422 \u043a\u0430\u043f\u0441.5\u043c\u0433*60",
         "quantity":1,
         "unitPrice":30,
         "taxGroup":2,
         "priceModifierValue":3.36,
         "priceModifierType":"discount-amount"
      },
      { 
         "text":"\u041f\u0417 \u0423\u041b\u0422\u0420\u0410\u0411\u041b\u0410\u041d\u041a\u0410 \u0418\u0417\u0411\u0415\u041b\u0412\u0410\u0429\u0410 75 \u043c\u043b",
         "quantity":1,
         "unitPrice":13.6,
         "taxGroup":2,
         "priceModifierValue":0.18,
         "priceModifierType":"discount-amount"
      },
      { 
         "text":"\u041f\u0430\u043a\u0435\u0442 \u201e\u041e\u0447\u043d\u043e \u0437\u0434\u0440\u0430\u0432\u0435\u201c \u043e\u0442 \u043e\u043f\u0442\u0438\u043a\u0438 Grand Optics & Joy Optics",
         "quantity":1,
         "unitPrice":0,
         "taxGroup":2
      },
      { 
         "text":"\u0414\u043e\u0441\u0442\u0430\u0432\u043a\u0430",
         "quantity":1,
         "unitPrice":4,
         "taxGroup":2
      }
   ],
   "payments":[ 
      { 
         "amount":44.06,
         "paymentType":"reserved1"
      }
   ]
}
```
debug.log
```
[2019-11-11 15:36:46 INF] Request starting HTTP/1.1 GET http://localhost:8001/printers/dt522935/status  
[2019-11-11 15:36:46 INF] CORS policy execution successful.
[2019-11-11 15:36:46 INF] -- HTTP Request -- GET: http://localhost:8001/printers/dt522935/status, Body(0):
[2019-11-11 15:36:46 INF] Executing endpoint 'ErpNet.FP.Server.Controllers.PrintersController.Status (ErpNet.FP.Server)'
[2019-11-11 15:36:46 INF] Route matched with {action = "Status", controller = "Printers"}. Executing controller action with signature Microsoft.AspNetCore.Mvc.ActionResult`1[ErpNet.FP.Core.DeviceStatusWithDateTime] Status(System.String) on controller ErpNet.FP.Server.Controllers.PrintersController (ErpNet.FP.Server).
[2019-11-11 15:36:46 INF] bg.dt.c.isl.com://COM1 <<< 01-24-27-3E-05-30-30-38-3E-03
[2019-11-11 15:36:46 INF] Opening the com port COM1
[2019-11-11 15:36:46 INF] bg.dt.c.isl.com://COM1 >>> 16
[2019-11-11 15:36:46 INF] bg.dt.c.isl.com://COM1 >>> 01
[2019-11-11 15:36:46 INF] bg.dt.c.isl.com://COM1 >>> 3C-27-3E-31-31-2D-31-31-2D-31-39-20-31-35-3A
[2019-11-11 15:36:46 INF] bg.dt.c.isl.com://COM1 >>> 33-38-3A-34-34-04-80-80
[2019-11-11 15:36:46 INF] bg.dt.c.isl.com://COM1 >>> 80-80-86-9A-05-30-37-31-3F-03
[2019-11-11 15:36:46 INF] Executing ObjectResult, writing value of type 'ErpNet.FP.Core.DeviceStatusWithDateTime'.
[2019-11-11 15:36:46 INF] Executed action ErpNet.FP.Server.Controllers.PrintersController.Status (ErpNet.FP.Server) in 11.0593ms
[2019-11-11 15:36:46 INF] Executed endpoint 'ErpNet.FP.Server.Controllers.PrintersController.Status (ErpNet.FP.Server)'
[2019-11-11 15:36:46 INF] -- HTTP Response -- Code: 200, Body(438):
{"deviceDateTime":"2019-11-11T15:38:44","ok":true,"messages":[{"type":"info","text":"SW7=OFF, SW6=OFF, SW5=OFF, SW4=OFF, SW3=OFF, SW2=OFF, SW1=OFF"},{"type":"info","text":"Unique Printer ID and Fiscal Memory ID are set"},{"type":"info","text":"BULSTAT UIC is set"},{"type":"info","text":"The tax rates are set at least once"},{"type":"info","text":"The printer is in fiscal mode"},{"type":"info","text":"The fiscal memory is formatted"}]}
[2019-11-11 15:36:46 INF] HTTP GET /printers/dt522935/status responded 200 in 19.3307 ms
[2019-11-11 15:36:46 INF] Request finished in 23.2907ms 200 application/json; charset=utf-8
[2019-11-11 15:36:46 INF] Request starting HTTP/1.1 OPTIONS http://localhost:8001/printers/dt522935/receipt?asyncTimeout=0  
[2019-11-11 15:36:46 INF] CORS policy execution successful.
[2019-11-11 15:36:46 INF] Request finished in 2.3802000000000003ms 204 
[... 38 lines cut ...]
   at Microsoft.AspNetCore.WebUtilities.HttpRequestStreamReader.Read(Char[] buffer, Int32 index, Int32 count)
   at Newtonsoft.Json.JsonTextReader.ReadData(Boolean append, Int32 charsRequired)
   at Newtonsoft.Json.JsonTextReader.ParsePostValue(Boolean ignoreComments)
   at Newtonsoft.Json.JsonTextReader.Read()
   at Newtonsoft.Json.Serialization.JsonSerializerInternalReader.PopulateObject(Object newObject, JsonReader reader, JsonObjectContract contract, JsonProperty member, String id)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalReader.CreateObject(JsonReader reader, Type objectType, JsonContract contract, JsonProperty member, JsonContainerContract containerContract, JsonProperty containerMember, Object existingValue)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalReader.CreateValueInternal(JsonReader reader, Type objectType, JsonContract contract, JsonProperty member, JsonContainerContract containerContract, JsonProperty containerMember, Object existingValue)
   at Newtonsoft.Json.Serialization.JsonSerializerInternalReader.Deserialize(JsonReader reader, Type objectType, Boolean checkAdditionalContent)
--- End of stack trace from previous location where exception was thrown ---
   at Microsoft.AspNetCore.Mvc.Formatters.NewtonsoftJsonInputFormatter.ReadRequestBodyAsync(InputFormatterContext context, Encoding encoding)
   at Microsoft.AspNetCore.Mvc.ModelBinding.Binders.BodyModelBinder.BindModelAsync(ModelBindingContext bindingContext)
   at Microsoft.AspNetCore.Mvc.ModelBinding.ParameterBinder.BindModelAsync(ActionContext actionContext, IModelBinder modelBinder, IValueProvider valueProvider, ParameterDescriptor parameter, ModelMetadata metadata, Object value)
   at Microsoft.AspNetCore.Mvc.Controllers.ControllerBinderDelegateProvider.<>c__DisplayClass0_0.<<CreateBinderDelegate>g__Bind|0>d.MoveNext()
--- End of stack trace from previous location where exception was thrown ---
   at Microsoft.AspNetCore.Mvc.Infrastructure.ControllerActionInvoker.<InvokeInnerFilterAsync>g__Awaited|13_0(ControllerActionInvoker invoker, Task lastTask, State next, Scope scope, Object state, Boolean isCompleted)
   at Microsoft.AspNetCore.Mvc.Infrastructure.ResourceInvoker.<InvokeFilterPipelineAsync>g__Awaited|19_0(ResourceInvoker invoker, Task lastTask, State next, Scope scope, Object state, Boolean isCompleted)
   at Microsoft.AspNetCore.Mvc.Infrastructure.ResourceInvoker.<InvokeAsync>g__Logged|17_1(ResourceInvoker invoker)
   at Microsoft.AspNetCore.Routing.EndpointMiddleware.<Invoke>g__AwaitRequestTask|6_0(Endpoint endpoint, Task requestTask, ILogger logger)
   at ErpNet.FP.Server.Middlewares.ActionLoggingMiddleware.Invoke(HttpContext context)
   at Serilog.AspNetCore.RequestLoggingMiddleware.Invoke(HttpContext httpContext)
   at Microsoft.AspNetCore.Server.Kestrel.Core.Internal.Http.HttpProtocol.ProcessRequests[TContext](IHttpApplication`1 application)
[2019-11-11 15:36:46 INF] Request finished in 32.7186ms 500 
[2019-11-11 15:36:47 INF] Idle timer elapsed for the com port COM1
[2019-11-11 15:36:47 INF] Closing the com port COM1
```

g-vvv, 2019-11-17:
It looks like Kestrel issue. Try patching this snipped in Program.cs file:
```
                .ConfigureKestrel((hostingContext, options) =>
                {
                    options.Configure(hostingContext.Configuration.GetSection("Kestrel"));

                    // Overriding some of the config values 
                    options.AllowSynchronousIO = true; // it was false
                    options.Limits.MaxRequestBodySize = 500 * 1024;
                })
```
The change is: ```options.AllowSynchronousIO = true;``` instead of ```options.AllowSynchronousIO = false;``` and if it works for you, ask for pull request to merge it with the master branch.

valeriterziev, 2019-11-18:
I have trouble with several reference errors on compile time:
Core\IFiscalPrinter.cs(33,10) error CS0246: The type or namespace name 'ReceiptInfo' could not be found (are you missing a using directive or an assembly reference?)

synapio, 2019-11-18:
Please explain to us your development environment, so we can help you.

valeriterziev, 2019-11-18:
I hope this will help:

**OS Name	Microsoft Windows 10 Home**
Version	10.0.18362 Build 18362

**Visual Studio Build Tools 2019 16.3.9**
.NET Core build tools 
.NET Core 3.0 SDK
NuGet targets and build tasks

**Microsoft Visual Studio Community 2019**
Version 16.3.9
VisualStudio.16.Release/16.3.9+29509.3
Microsoft .NET Framework
Version 4.8.03752

ASP.NET and Web Tools 2019   16.3.286.43615
Azure App Service Tools v3.0.0   16.3.286.43615
C# Tools   3.3.1-beta3-19461-02+2fd12c210e22f7d6245805c60340f6a34af6875b
Common Azure Tools   1.10
IntelliCode Extension   1.0
Microsoft Continuous Delivery Tools for Visual Studio   0.4
Microsoft JVM Debugger   1.0
Microsoft Library Manager   2.0.83+gbc8a4b23ec
Microsoft MI-Based Debugger   1.0
Microsoft Visual Studio Tools for Containers   1.1
NuGet Package Manager   5.3.1
ProjectServicesPackage Extension   1.0
SQL Server Data Tools   16.0.61908.27190
TypeScript Tools   16.0.10821.2002
Visual Basic Tools   3.3.1-beta3-19461-02+2fd12c210e22f7d6245805c60340f6a34af6875b
Visual F# Tools 10.4 for F# 4.6   16.3.0-beta.19455.1+0422ff293bb2cc722fe5021b85ef50378a9af823
Visual Studio Code Debug Adapter Host Package   1.0
WiX Toolset Visual Studio Extension   1.0.0.4

g-vvv, 2019-11-18:
Run from command line:
``` dotnet restore ```
inside the solution's and every project's folders, to make sure that all dependencies are satisfied.

g-vvv, 2019-11-18:
You can clear all ```bin``` and ```obj``` folders for clean up before rebuilding the solution.
After that, again from the console inside the solution's folder ErpNet.FP, run:
``` dotnet msbuild output.xml ```
to build all projects and to store them inside the ```Publish``` and ```Output``` folders inside the solution's folder.

valeriterziev, 2019-11-18:
Yes, I had done all this before.
Excuse losing your time, I am not C# developer and probably some trivial missing in my environment.

D:\ErpNet.FP>dotnet restore
  Restore completed in 204.45 ms for D:\ErpNet.FP\ErpNet.FP.Core\ErpNet.FP.Core.csproj.
  Restore completed in 332.11 ms for D:\ErpNet.FP\ErpNet.FP.Server\ErpNet.FP.Server.csproj.

D:\ErpNet.FP>dotnet msbuild output.xml
Microsoft (R) Build Engine version 16.3.0+0f4c62fea for .NET Core
Copyright (C) Microsoft Corporation. All rights reserved.

  Performing cleanup...
  Publishing binaries for win-x86 into Published folder...
  Publishing binaries for win-x64 into Published folder...
  Publishing binaries for osx-x64 into Published folder...
  Publishing binaries for linux-x64 into Published folder...
  Publishing binaries for linux-arm into Published folder...
  Microsoft (R) Build Engine version 16.3.0+0f4c62fea for .NET Core
  Copyright (C) Microsoft Corporation. All rights reserved.

Core\DeviceStatus.cs(144,65): error CS0246: The type or namespace name 'ReceiptInfo' could not be found (are you missing a using directive or an assembly reference?)
...

synapio, 2019-11-18:
We cannot reproduce your error. Maybe your directory is not clean.
We make another folder, clean:

synapio, 2019-11-18:
We cannot reproduce your error. Maybe your directory is not clean.
Just make another folder, clean:
```
C:\Users\user\go\src\github.com\test>git clone https://github.com/erpnet/ErpNet.FP
Cloning into 'ErpNet.FP'...
remote: Enumerating objects: 74, done.
remote: Counting objects: 100% (74/74), done.
remote: Compressing objects: 100% (60/60), done.
remote: Total 4650 (delta 37), reused 33 (delta 14), pack-reused 4576
Receiving objects: 100% (4650/4650), 439.96 MiB | 11.88 MiB/s, done.
Resolving deltas: 100% (3184/3184), done.

C:\Users\user\go\src\github.com\test>cd ErpNet.FP

C:\Users\user\go\src\github.com\test\ErpNet.FP>dotnet msbuild output.xml
Microsoft (R) Build Engine version 16.3.0+0f4c62fea for .NET Core
Copyright (C) Microsoft Corporation. All rights reserved.

  Performing cleanup...
  Publishing binaries for win-x86 into Published folder...
  Publishing binaries for win-x64 into Published folder...
  Publishing binaries for osx-x64 into Published folder...
  Publishing binaries for linux-x64 into Published folder...
  Publishing binaries for linux-arm into Published folder...
  Microsoft (R) Build Engine version 16.3.0+0f4c62fea for .NET Core
  Copyright (C) Microsoft Corporation. All rights reserved.

  Microsoft (R) Build Engine version 16.3.0+0f4c62fea for .NET Core
  Copyright (C) Microsoft Corporation. All rights reserved.

  Microsoft (R) Build Engine version 16.3.0+0f4c62fea for .NET Core
  Copyright (C) Microsoft Corporation. All rights reserved.

  Microsoft (R) Build Engine version 16.3.0+0f4c62fea for .NET Core
  Copyright (C) Microsoft Corporation. All rights reserved.

  Microsoft (R) Build Engine version 16.3.0+0f4c62fea for .NET Core
  Copyright (C) Microsoft Corporation. All rights reserved.

  Zipping binaries for all supported target platforms into Output folder...
  Zipping directory "C:\Users\user\go\src\github.com\test\ErpNet.FP\Published\win-x86" to "C:\Users\user\go\src\github.com\test\ErpNet.FP\Output\win-x86.zip".
  Zipping directory "C:\Users\user\go\src\github.com\test\ErpNet.FP\Published\win-x64" to "C:\Users\user\go\src\github.com\test\ErpNet.FP\Output\win-x64.zip".
  Zipping directory "C:\Users\user\go\src\github.com\test\ErpNet.FP\Published\osx-x64" to "C:\Users\user\go\src\github.com\test\ErpNet.FP\Output\osx-x64.zip".
  Zipping directory "C:\Users\user\go\src\github.com\test\ErpNet.FP\Published\linux-x64" to "C:\Users\user\go\src\github.com\test\ErpNet.FP\Output\linux-x64.zip".
  Zipping directory "C:\Users\user\go\src\github.com\test\ErpNet.FP\Published\linux-arm" to "C:\Users\user\go\src\github.com\test\ErpNet.FP\Output\linux-arm.zip".
  Everything is ready. Now you could run rebuild of ErpNet.FP.Setup to create the installer into Output folder
```
and as you can see, everything compiles.

g-vvv, 2019-11-18:
Yes, @valeriterziev, try with clean folder, and fresh git clone. After that, just make the patch, and try to run the ErpNet.FP.Server. If it solves this issue, report it here.

valeriterziev, 2019-11-18:
Thank you for help. Fresh git clone solve the compilation.
I confirm `options.AllowSynchronousIO = true;` solve the issue.

Seems error is caused by "&" character in one of the items. But only in some specific combination with other items/letters. The same item is printed in around 9000 receipts, and only around 10 receipts are reported with this error.

g-vvv, 2019-11-18:
Ok, thanks for the valuable information. I will make the patch and will publish new release in the next schedule - at the end of this week (we have more things that will be changed until then). But this strange case with ampersand symbol inside the body of the request makes me wonder, about Kestrel implementation :)

g-vvv, 2019-11-22:
New version released.

closed 2019-11-22 by g-vvv