erpnet/ErpNet.FP #122: Accessing the API outside of localhost

dmihaylov92, 2019-11-13:
Hello, first to say great work on the API. It's really easy to work with.
I'm integrating a fiscal device with a web based system (PHP) and I'd like to use your API to communicate via the device. I've installed ErpNET.FP and it works fine on localhost, but I'm unable to access the API from another pc that is also in the same network using the hosts IP and port.

I've searched around for similar issues and people are suggesting to create reverse proxy via IIS or nginx, apache... Is this the best solution? Is it okey to use only Kestrel for such purposes and can I setup it to be accessed locally/publicly? 
 
Kestrel config:
`
"Kestrel": {
    "EndPoints": {
      "Http": {
        "Url": "http://0.0.0.0:8001"
      }
    }
}
`

I've also tried to expose the in windows firewall, but no change.
Regards and thanks in advance

dmihaylov92, 2019-11-13:
All good, exposing the host's port fixed the problem.
Still, because I'm not a C# Developer, I'd like to know for your opinion.
What would you suggest, should I create reverse proxy on a API host machine, or it's okey to use Kestrel-only. 

Thanks!

g-vvv, 2019-11-17:
"Kestrel": { "EndPoints": { "Http": { "Url": "http://localhost:8001" } } } will prevent accessing it outside from localhost

closed 2019-11-17 by g-vvv