erpnet/ErpNet.FP #125: Add ability to change speed for each printer

dmihaylov92, 2019-11-21:
From [BGFiscalPrinterSetup.md](https://github.com/erpnet/ErpNet.FP/blob/master/BGFiscalPrinterSetup.md)

> ВНИМАНИЕ: Библиотеката очаква серийният порт на фискалното устройство (ФУ) да е конфигуриран за скорост 115200 bps. Повечето ФУ са настроени за тази скорост, но ако това не е така, то може да го конфигурирате със сервизната му програма или в оторизирания сервиз.

Hello guys, first, great and quick support on the library.
I'd like to ask, would it be possible to set the speed for each fiscal device/printer when we are adding it manually?

Or another alternative would be if we are able to change the value in configuration json.
I would be nice if we are able to change the value of timeout too (currently 800ms). 

Apologies if there's already a way to override those values. Unfortunately I was not able to find them in docs or closed issues.

Best regards, Denislav

g-vvv, 2019-11-21:
The Speed or the Timeout? There is a big difference. 
You can set the speed from the URI, for example: bg. dt.c.isl://COM5:9600

g-vvv, 2019-11-21:
The timeout is good idea, but overriding the defaults can cause an unexpected behavior in some of the devices. I am afraid that it will triger more issues and we will be forced to limit that parameter with lower and upper bounds. What is the use case, in which you want manually to change the timeout?

closed 2019-11-21 by g-vvv