erpnet/ErpNet.FP #126: An endpoint to return the number of the last receipt

f1mp3r, 2019-11-26:
**Is your feature request related to a problem? Please describe.**
The problem is described here: https://www.facebook.com/groups/BgBusinessDev/permalink/1396277443868116/

**Describe the solution you'd like**
If there is an endpoint which returns the number of the last printed document I would be able to just call it after each operation and check against saved ones in the database. If it exists, the print was not successful. If not - we have the last number and will save it even though the task returned an error and no number.

i-argentinski, 2019-11-26:
See documentation for idempotency:
https://github.com/erpnet/ErpNet.FP/blob/master/PROTOCOL.md#post-print-fiscal-receipt-idempotent

closed 2019-11-26 by i-argentinski

f1mp3r, 2019-11-26:
I am using this. The problem is different - a receipt is printed and added to the fiscal device balance/memory, but the response it returns is an error. This happens rarely (no more than once a month), and is a problem we've had with 2 different Datecs devices (DP-25 and FP-700X). So on our end, the receipt printing failed, but in reality it did not and we shouldn't try another print, because that would lead to difference between the server and the fiscal device.

synapio, 2019-11-26:
When it happens again, please provide debug.log here.

f1mp3r, 2019-11-26:
I found the log:
With idempotency, returned error, no receipt number: https://pastebin.com/FUNWP43h

synapio, 2019-11-26:
So, it is connection error. Use better usb to com cables or active usb hub
with power supply to avoid that error.

g-vvv, 2019-11-26:
When there is such error, you should support manual entering the number of the receipt, the amount, the date time of the receipt and fiscal serial number of the device from the paper receipt printout. Reading last receipt info will be NOT implemented, because it is not transaction safe. If the device is used in network, and there are multiple users, sending receipts to the device, or there is a another, or many other printing jobs in the queue, last receipt info maybe will be another job's receipt information. It is not safe to use the last receipt info in multiuser environment. ErpNet.FP is created to be used in multiuser environments, so allowing this functionality in long run will be error prone.