erpnet/ErpNet.FP #127: DP-150 - The command is not allowed in the current fiscal mode

dmihaylov92, 2019-12-02:
Hello, today I tried to open a receipt on a fiscalized device and I'm receiving this error:

The command is not allowed in the current fiscal mode

The fiscal device is set on PC Mode.
Tried also to reset it from the button "Reset" in API Interface.

[debug.log](https://github.com/erpnet/ErpNet.FP/files/3910402/debug.log)

Any ideas where could be the issue.

Thank you for your support.

iveivanov, 2019-12-02:
Hello,
Try to print Z-report first.

dmihaylov92, 2019-12-02:
Hi @iveivanov , thank you for getting back to me.
We've tried and the issue persists.

I'm attaching the new debug.log (due system restart)
[debug.log](https://github.com/erpnet/ErpNet.FP/files/3911149/debug.log)

The request starts at "2019-12-02 13:05:09"

Any ideas. I've doubled check the json that I'm sending but I'm not able to see any issues with it.

Also we've changed the default communication speed value of ERP API to 38400, because the fiscal devices are set to that speed (this was made before I knew how to set the speed when manually adding a device), could this be the reason of the issue?

g-vvv, 2019-12-02:
You've made a mistake, the name of the array it is not "payment", it is "payments", because you can pay with multiple methods - payment types, in one sale - fiscal receipt.
Please use the RESET button on the ErpNet.FP Admin Page and try again. Be sure that you've aborted the erroneus receipt.

closed 2019-12-02 by g-vvv

dmihaylov92, 2019-12-03:
Hello, thank you for getting back to me. Unfortunately it seems that it didn't work.
What I did:
1. I corrected json data that I'm sending. 
2. Reset the fiscal device from the interface button "reset"
3. Tried to create new receipt - same issue.

[2019-12-03 11:57:48 INF] - Reset request
[2019-12-03 11:58:05 INF] - Receipt request

JSON Request:
`{
   "uniqueSaleNumber":"DT742302-0001-0000001",
   "items":[
      {
         "text":"\u0440\u0438\u0442. \u043a\u043d.",
         "quantity":1,
         "unitPrice":6,
         "taxGroup":1
      }
   ],
   "payments":[
      {
         "amount":6,
         "paymentType":"cash"
      }
   ]
}`

JSON Response:
`{
   "receiptNumber":"",
   "receiptDateTime":"0001-01-01T00:00:00",
   "receiptAmount":0.0,
   "fiscalMemorySerialNumber":"",
   "ok":false,
   "messages":[
      {
         "type":"error",
         "code":"E199",
         "text":"General error"
      },
      {
         "type":"error",
         "code":"E404",
         "text":"The command is not allowed in the current fiscal mode"
      },
      {
         "type":"info",
         "text":"SW7=OFF, SW6=OFF, SW5=OFF, SW4=OFF, SW3=OFF, SW2=OFF, SW1=OFF"
      },
      {
         "type":"info",
         "text":"Unique Printer ID and Fiscal Memory ID are set"
      },
      {
         "type":"info",
         "text":"BULSTAT UIC is set"
      },
      {
         "type":"info",
         "text":"The tax rates are set at least once"
      },
      {
         "type":"info",
         "text":"The printer is in fiscal mode"
      },
      {
         "type":"info",
         "text":"The fiscal memory is formatted"
      },
      {
         "type":"info",
         "text":"Error occured while opening new fiscal receipt"
      }
   ]
}`

I'm not very familiar with fiscal devices, I see that there's a lot of "info" warnings, are they normal, could some of them be the issue? 

Also, we are working with another program for receipts that is using datecs FPrint and the device is printing a receipt without any problems. 
The channel itself is open to use. (we are not trying to print 2 receipts at the same time from both programs)
After the last night's Z-Report, today we tried to print a receipt first via Erp API (after the changes to json request), and it didn't let us, but is it possible if we print a receipt via Program A's FPrint and try to print a receipt after that via ERP to don't let us.

Just now double checked, the device is in PC Mode.  

[debug.log](https://github.com/erpnet/ErpNet.FP/files/3915995/debug.log)

Apologies for the inconvenience, but I'm out of ideas and don't know what else to try. 
Tried also with floating zeros for prices, no change.

synapio, 2019-12-03:
Check your device speed, and set it to 115200.

g-vvv, 2019-12-03:
So, you use two libraries that connects simultaneously to the fiscal device, ErpNet.FP and another serial printing library in the same time. This could lead to your problems. Avoid this behaviour while using the ErpNet.FP.

dmihaylov92, 2019-12-04:
Hello everyone, first, I'd like to thank you for your support.
At the end we established that we need to provide the operator and operator password. (only when priting receipts), had no such issue with printing x and z reports.  

So if anybody has error message E404 - This command is not allowed in the current fiscal mode, double/triple check your json data.

Thank you again for your support

g-vvv, 2019-12-04:
I want to thank you for finding another reason why the receipt could not be printed. We will keep it in mind when someone else encounters the same problem. 
In doing so, you have helped many more ErpNet.FP users.
So, thank You :)