erpnet/ErpNet.FP #130: Add to printer config  itemTextLength and commentTextLength

iveivanov, 2019-12-12:
It is a good idea to have possibility to define itemTextLength and commentTextLength in printer definition, because printer documentation is not correct and ErpNet.FP doesn't return proper values.
This can be a config file then can be updated remotely or locally.

iveivanov, 2019-12-13:
Hello,

Thank you for the new settings! 
But there is a problem with them. After I set the commentTextLength and apply, the server returned original values:
[image]

[image]

g-vvv, 2019-12-13:
There is new version, v.1.0.1213.1913. Try with it.

closed 2019-12-14 by g-vvv